litex-hub/litex-boards #290: Tang nano internal uart chip

developandplay, 2021-11-06:
As a follow up to https://github.com/litex-hub/litex-boards/pull/280 I tried getting uartbone to work on the tang nano.
Using an external ftdi chip this works as expected.
However using the internal ch552 uart chip I didn't get it to work.
Other projects mention (like this one https://github.com/kingyoPiyo/Tang-Nano_I2C_Monitor) that the firmware of the ch552 needs to be updated first. However that alone didn't do the trick for me.
As loopback is working using the ch552 I'm a bit confused as to what could be interferring with the uart communication.

enjoy-digital, 2021-11-08:
Hi @developandplay,

I just updated the CH552 firmware on my Tang Nano and reproduced the issue you had (working loopback/failing UARTBone) and got it with https://github.com/litex-hub/litex-boards/commit/184f41e61a4d0d0e7e1060485ff852cbcd727ab9.

There seems to be a couple of quirks:
- The CH522 firmware will only works correctly at some baudrates, 1MBauds was the one used on other tutorials and seems to works correctly.
- The CH522 firmware seems to require receiving a few bytes from the Host before operating correctly, so I patched litex_server for this. 

With this, the UARTBone bridge operate correctly (tested with litex_cli --write/--read//--regs). Could you also do a test on your side and tell me if you reproduce these results?

developandplay, 2021-11-08:
Awesome work!
That works for me as well. :partying_face:

closed 2021-11-08 by developandplay

enjoy-digital, 2021-11-09:
Great, thanks for the feedback.