litex-hub/litex-boards #316: BIOS not starting on GOWIN Trenz TEC0117

hubmartin, 2021-12-28:
I sucesfully generated gateware, flashed it with a little fix in #315 . The LED chaser works but I don't see any output on UART. So it seems like something is wrong with BIOS.
When I used Trenz demo bitstream, then I get some output on UART, so USB to UART converter works fine.

Any ideas how to troubleshoot this are welcome. Thanks

enjoy-digital, 2022-01-03:
Hi @hubmartin,

I just did a test and got it working with` ./trenz_tec0117.py --build --flash`:

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2021 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jan  3 2022 17:32:35
 BIOS CRC passed (265df2a8)

 Migen git sha1: ac70301
 LiteX git sha1: 3b05fa57

--=============== SoC ==================--
CPU:		VexRiscv @ 25MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		64KiB
SRAM:		8KiB
L2:		0KiB
FLASH:		8192KiB
SDRAM:		8192KiB 16-bit @ 25MT/s (CL-2 CWL-2)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 6.1MiB/s
   Read speed: 6.3MiB/s

Initializing W74M64FV SPI Flash @0x80000000...
SPI Flash clk configured to 12 MHz
Memspeed at 0x80000000 (Sequential, 4.0KiB)...
   Read speed: 2.9MiB/s
Memspeed at 0x80000000 (Random, 4.0KiB)...
   Read speed: 526.0KiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
No boot medium found

--============= Console ================--

litex> 
```

`--flash` seems to mess up FTDI2232H, so once flashed, you have to disconnect/reconnect the board and you can then use litex_term on the second UART: `litex_term /dev/ttyUSB1` if FT2232 creates `/dev/ttyUSB0` and `/dev/ttyUSB1`.

hubmartin, 2022-01-03:
It is working!
I guess I had the issue with the FTDI beign reconfigured. Without reconfiguration I could see only one serial port. After USB reconnect both ttyUSBs appeared and the second one welcomed me with console.

Thanks a lot for your help. Litex is really awesome project so that with minimal FPGA/gateware knowledge, you can create a really complex system. :+1:

closed 2022-01-03 by hubmartin