litex-hub/litex-boards #329: CLE-215 DDR RAM initialization failture

zeroday-engineer, 2022-01-19:
Hi,

I tried following the SQRL Acorn project and building a bitstream specifically for the CLE-215 using 

`python3 -m litex_boards.targets.sqrl_acorn --variant=cle-215 --with-pcie --driver  --doc --uart-name=crossover --uart-baudrate=9600 --uart-fifo-depth=16`

I'm able to program my device without issue, and run the basic utilities provided, but noticed when I output the UART from the device, it seems the DDR3 ram failed to initialize. Is this a known issue?

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jan 18 2022 21:18:26
 BIOS CRC passed (f2f6bdf5)

 Migen git sha1: ac70301
 LiteX git sha1: add2746a

--=============== SoC ==================--
CPU:		VexRiscv @ 100MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128KiB
SRAM:		8KiB
L2:		8KiB
SDRAM:		1048576KiB 16-bit @ 800MT/s (CL-7 CWL-5)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b00 delays: -
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b00 delays: -
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
  bus errors:  256/256
  addr errors: 8191/8192
  data errors: 524288/524288
Memtest KO
Memory initialization failed

--============= Console ================--

zeroday-engineer, 2022-01-20:
Tried going as low as 25Mhz for the sys-clk, just to see if the DDR issues were related to timing/clk issues, still got the same results. 

Also tried femtorv as the CPU type, and got the same result as well

enjoy-digital, 2022-01-20:
@zeroday-engineer: The DDR3 is working on the CLE215+ yes but I never tested the CLE215. The only difference seems to be the speedgrade (-2 instead of -3). You can attached the bitstream I used on the  [Acorn giveway campaign](https://twitter.com/enjoy_digital/status/1461353848388284422).

[acorn_pcie_cle215+.zip](https://github.com/litex-hub/litex-boards/files/7906373/acorn_pcie_cle215%2B.zip)

If it does not work, it would be interesting to verify that the only difference is only the speedgrade and if you have another board to test with it since it could be an hardware issue.

zeroday-engineer, 2022-01-20:
@enjoy-digital It does appear that I have a bad board, tried the default build on a second device (should've thought of that earlier) and DDR Ram got setup without issue

closed 2022-01-21 by zeroday-engineer