litex-hub/litex-boards #349: digilent_nexys4[ddr] targets don't support the --toolchain option and therefore can't use SymbiFlow

tcal-x, 2022-02-10:
Is there an underlying reason they can't use SymbiFlow, or is this just an oversight?   

If it's just an oversight, we should look through the rest of the boards that use xc7a devices to see if any of them are also missing the toolchain option.

tcal-x, 2022-02-14:
`digilent_basys3.py` is also missing the `--toolchain` option.

enjoy-digital, 2022-02-14:
Hi @tcal-x,

only supporting SymbiFlow on the Arty was an initial wanted limitation. Now that Symbiflow supports more xc7a devices, we could indeed extend it and eventually have this directly integrated in LiteX.

enjoy-digital, 2022-02-15:
@tcal-x: To prepare this, I already made sure the toolchain parameter was exposed on all platforms throug: https://github.com/litex-hub/litex-boards/commit/b53b79821e757a92ed9b7f15d7ebe5ad7a0bbd44. I'll try to go further in the next days.

tcal-x, 2022-02-15:
Thank you @enjoy-digital .  It is not urgent, but it seemed to be a gap that would be good to fill in when one of us finds the time.

enjoy-digital, 2026-10-21:
All targets now expose `--toolchain` through the common LiteX argument parser (`parser.toolchain_argdict`), including `digilent_nexys4`, `digilent_nexys4ddr` and `digilent_basys3`, so closing. Thanks!

closed 2026-10-21 by enjoy-digital