litex-hub/litex-boards #352: Common PMODs and peripherals

suarezvictor, 2022-02-20:
I integrated DVI output in the de0nano board with a custom pmod, and VGA output on the arty using the standard pmod.
However in the board descriptions of this repo there is no support

Should we add pin descriptions of common or useful peripherals? In the case of de0 nano, dvi output uses LVDS pins that are in specific locations so most users will follow the configuration.

The idea is that if the switch --with-video-framebuffer is set in that boards (and possibility others) it works out of the box.

Is this aligned with the current design? Or the current design is just to support peripherals that comes integrated in the board?

In the first case I can send the patches

enjoy-digital, 2022-02-22:
Hi @suarezvictor,

the idea is to avoid over-engineering, so you can do something similar to what is done on Digitlent Arty: 
https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/digilent_arty.py#L251-L333

We could improve this in the future when more boards will use these features and when we'll know better the common integration patterns.

enjoy-digital, 2022-05-27:
As described in previous message, I would recommend for now to use a similar approach than the one used for example on the Digilent Arty (and probably others boards). We could generalize this in the future if useful, for now this seems premature.

closed 2022-05-27 by enjoy-digital