litex-hub/litex-boards #358: How to build/load for Digilent Arty Z7 (Zynq, xc7z010)

tcal-x, 2022-03-01:
Hello, I've had difficulty finding a basic recipe to build and load for Arty Z7 (and also, I think I found a minor bug in the platform file).

The bug first:  I think this line should have `1e9`, not `1e6`:
```
    self.add_period_constraint(self.lookup_request("clk125", loose=True), 1e6/125e6)
```

My first attempt was just `./digilent_arty_z7.py --build --load`, which gives:
```
ERROR:SoC:CPU needs reset address 0xfc000000 to be in a defined Region.
ERROR:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
IO Regions: (1)
io0                 : Origin: 0x40000000, Size: 0xbc000000, Mode: RW, Cached: False Linker: False
Bus Regions: (1)
csr                 : Origin: 0x40000000, Size: 0x00010000, Mode: RW, Cached: False Linker: False
Bus Masters: (1)
- master0
Bus Slaves: (1)
- csr
```

Then I realized that the default cpu type was zynq7000, which I don't need to deal with, so I then tried:
```
./digilent_arty_z7.py --build --load --cpu-type=vexriscv
```
which did complete, and seemed to put a bitstream on the board, but I don't see the LEDChaser working, and I can't connect over serial.

I'm guessing that the USB port on the board is connected to the PS (ARM) and not the PL, so if I want to connect to the VexRiscv, I will need to use a UART/USB converter pmod.   But I'm puzzled by the lack of LEDChaser.

enjoy-digital, 2022-03-01:
Hi @tcal-x,

I just fixed the period constraints and removed the fake serial. 

When building the design with the following command I get the LedChaser: 
`./digilent_arty_z7.py --cpu-type=vexriscv --uart-name=stub --build --load`

Can you re-do the test?

Otherwise, for the UART, it's indeed not possible to use the one connected to the FTDI since directly connected to the PS. You can add a UART PMOD to the board for now or on the future we should be able to use JTAG-UART as we are doing on other 7-Series devices, but this hasn't been validated yet.

tcal-x, 2022-03-01:
Thanks @enjoy-digital ,

I do get LEDChaser now, after getting past an error on my part -- I was assuming the default variant was z7-10 (which is what I want), but the default is z7-20 when a soft CPU is specified.   Once I added `--variant=z7-10`, I was able to get the LEDChaser.

I'll close this, thanks again!

closed 2022-03-01 by tcal-x