litex-hub/litex-boards #364: ECP5 Versa Board PCIe

RArbore, 2022-03-11:
Hello,

I'm wondering whether the ECP5 Versa evaluation board is supported via Litex w/ PCIe? I see that it is on the board list for this repo, but I don't see it mentioned anywhere on the litepcie repo - there is an issue related to it, but it didn't seem like it got resolved (https://github.com/enjoy-digital/litepcie/issues/20). If not, is the fully open toolchain for Artix 7 chips mature enough to the point where I could create a PCIe design with an Artix 7 board using just open tools?

enjoy-digital, 2022-03-14:
Hello @RArbore,

LitePCIe is not supported on the ECP5 Versa (as you saw due to the lack of PCIe PHY). The F4PGA team has already been able to use GTP transceivers with Xilinx Artix7 (https://antmicro.com/blog/2021/12/sata-with-open-source-fpga-tools/) but integrating PCIe was one of the next plans IIRC. So this is not yet available but is not too unrealistic in a near future.

Another future option for PCIe could be the Lattice Certus-NX versa board that should have good open-source toolchain suport and hardened PCIe PHY. It's possible that LitePCIe support will be added to these devices in a near future.

enjoy-digital, 2022-03-16:
The Xilinx files are mostly wrappers around the Transceiver and PCIe Hard block. Some equivalent would need to be created if you don't want to use them. This is not something we planned but the effort could be evaluated if you want to fund such work. If F4FPGA add PCIe support, it will probably by reusing the Xilinx files.

enjoy-digital, 2022-03-24:
The question has been answered, closing.

closed 2022-03-24 by enjoy-digital