litex-hub/litex-boards #368: qmtech_ep4cex5.py build with ethernet failed

yjdwbj, 2022-03-14:
Hi All. I'm execute follow command got many errors.

```sh
litex-boards/litex_boards$ ./targets/qmtech_ep4cex5.py --with-ethernet --with-daughterboard --build

[...]
   Info (176467): Node "sdram_dq[15]" is constrained to location PIN V11 to improve DDIO timing File: /home/michael/workspace-xilinx/RISC-V/litex-hub/litex/litex-boards/litex_boards/build/qmtech_ep4cex5/gateware/qmtech_ep4cex5.v Line: 32
Info (176233): Starting register packing
Info (176235): Finished register packing
    Extra Info (176218): Packed 32 registers into blocks of type Block RAM
    Extra Info (176218): Packed 132 registers into blocks of type Embedded multiplier output
Info (171121): Fitter preparation operations ending: elapsed time is 00:00:31
Info (14896): Fitter has disabled Advanced Physical Optimization because it is not supported for the current family.
Info (170189): Fitter placement preparation operations beginning
Error (170040): Can't place all RAM cells in design
    Info (170034): Selected device has 56 memory locations of type M9K. The current design requires 78 memory locations of type M9K to successfully fit.
    Info (170033): Memory usage required for the design in the current device: 139% M9K memory block locations required
Info (170190): Fitter placement preparation operations ending: elapsed time is 00:00:01
Info (11888): Total time spent on timing analysis during the Fitter is 0.55 seconds.
Error (171000): Can't fit design in device
Warning (169177): 35 pins must meet Intel FPGA requirements for 3.3-, 3.0-, and 2.5-V interfaces. For more information, refer to AN 447: Interfacing Cyclone IV E Devices with 3.3/3.0/2.5-V LVTTL/LVCMOS I/O Systems.
    Info (169178): Pin eth_clocks_gtx uses I/O standard 3.3-V LVTTL at U22 File: /home/michael/workspace-xilinx/RISC-V/litex-hub/litex/litex-boards/litex_boards/build/qmtech_ep4cex5/gateware/qmtech_ep4cex5.v Line: 35
[.....]

Info (144001): Generated suppressed messages file /home/michael/workspace-xilinx/RISC-V/litex-hub/litex/litex-boards/litex_boards/build/qmtech_ep4cex5/gateware/qmtech_ep4cex5.fit.smsg
Error: Quartus Prime Fitter was unsuccessful. 2 errors, 24 warnings
    Error: Peak virtual memory: 1710 megabytes
    Error: Processing ended: Mon Mar 14 11:52:19 2022
    Error: Elapsed time: 00:01:02
    Error: Total CPU time (on all processors): 00:01:05
Info: *******************************************************************
Info: Running Quartus Prime Assembler
    Info: Version 20.1.1 Build 720 11/11/2020 SJ Lite Edition
    Info: Copyright (C) 2020  Intel Corporation. All rights reserved.
    Info: Your use of Intel Corporation's design tools, logic functions 
    Info: and other software and tools, and any partner logic 
    Info: functions, and any output files from any of the foregoing 
    Info: (including device programming or simulation files), and any 
    Info: associated documentation or information are expressly subject 
    Info: to the terms and conditions of the Intel Program License 
    Info: Subscription Agreement, the Intel Quartus Prime License Agreement,
    Info: the Intel FPGA IP License Agreement, or other applicable license
    Info: agreement, including, without limitation, that your use is for
    Info: the sole purpose of programming logic devices manufactured by
    Info: Intel and sold by Intel or its authorized distributors.  Please
    Info: refer to the applicable agreement for further details, at
    Info: https://fpgasoftware.intel.com/eula.
    Info: Processing started: Mon Mar 14 11:52:20 2022
Info: Command: quartus_asm --read_settings_files=off --write_settings_files=off qmtech_ep4cex5 -c qmtech_ep4cex5
Error (120000): Run Fitter (quartus_fit) before Assembler (quartus_asm)
Error: Quartus Prime Assembler was unsuccessful. 1 error, 0 warnings
    Error: Peak virtual memory: 352 megabytes
    Error: Processing ended: Mon Mar 14 11:52:20 2022
    Error: Elapsed time: 00:00:00
    Error: Total CPU time (on all processors): 00:00:00
Info: *******************************************************************
Info: Running Quartus Prime Timing Analyzer
    Info: Version 20.1.1 Build 720 11/11/2020 SJ Lite Edition
    Info: Copyright (C) 2020  Intel Corporation. All rights reserved.
    Info: Your use of Intel Corporation's design tools, logic functions 
    Info: and other software and tools, and any partner logic 
    Info: functions, and any output files from any of the foregoing 
    Info: (including device programming or simulation files), and any 
    Info: associated documentation or information are expressly subject 
    Info: to the terms and conditions of the Intel Program License 
    Info: Subscription Agreement, the Intel Quartus Prime License Agreement,
    Info: the Intel FPGA IP License Agreement, or other applicable license
    Info: agreement, including, without limitation, that your use is for
    Info: the sole purpose of programming logic devices manufactured by
    Info: Intel and sold by Intel or its authorized distributors.  Please
    Info: refer to the applicable agreement for further details, at
    Info: https://fpgasoftware.intel.com/eula.
    Info: Processing started: Mon Mar 14 11:52:20 2022
Info: Command: quartus_sta qmtech_ep4cex5 -c qmtech_ep4cex5
Info: qsta_default_script.tcl version: #1
Error: Can't run the Timing Analyzer (quartus_sta) -- Fitter (quartus_fit) failed or was not run. Run the Fitter (quartus_fit) successfully before running the Timing Analyzer (create_timing_netlist).
Error: Quartus Prime Timing Analyzer was unsuccessful. 1 error, 0 warnings
    Error: Peak virtual memory: 336 megabytes
    Error: Processing ended: Mon Mar 14 11:52:21 2022
    Error: Elapsed time: 00:00:01
    Error: Total CPU time (on all processors): 00:00:00

```

enjoy-digital, 2022-03-14:
Hi @yjdwbj,

an FPGA has limited resources and here Quartus indicates that the design is too big to fit in the device.  I just tried to reduce the resources  but the device seems too small to fit a SoC with CPU and Ethernet. BTW the `qmtech_ep4cex5` targs use the ep4ce15 FPGA device by default, but if you board uses a EP4CE55, you can add `--variant=ep4ce55` to your build command, the EP5E55 is a lot bigger and will probably have no trouble fitting this configuration.

enjoy-digital, 2022-03-24:
The initial question has  been answered, closing.

closed 2022-03-24 by enjoy-digital