litex-hub/litex-boards #381: GOWIN GW1NR-LV9 mem init + memtest failure

axibo-reiner, 2022-03-30:
Using the target for the tang nano 9k with no modifications. 

made with 

```bash
litex_boards.targets.sipeed_tang_nano_9k --build --flash --cpu-type=vexriscv --cpu-variant lite
```

```bash
 Migen git sha1: ac70301
 LiteX git sha1: aa3506a3

--=============== SoC ==================--
CPU:		VexRiscv_Lite @ 13MHz
BUS:		AXI-LITE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		64KiB
SRAM:		8KiB
FLASH:		4096KiB
MAIN-RAM:	4096KiB 

--========== Initialization ============--
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
  bus errors:  256/256
  addr errors: 0/8192
  data errors: 524288/524288
Memtest KO
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 6.4MiB/s
   Read speed: 2.0MiB/s
Memory initialization failed

Initializing W25Q32 SPI Flash @0x00000000...
SPI Flash clk configured to 6 MHz
Memspeed at 0 (Sequential, 4.0KiB)...
   Read speed: 700.2KiB/s
Memspeed at 0 (Random, 4.0KiB)...
   Read speed: 258.7KiB/s

--============= Console ================--
```
**What I have tested**
- reduced system clock frequency (13.5)
- axi-lite & wishbone
- have tried minimal, lite and full cpu. 

I am using the latest Gowin toolchain.

Now i am taking some time to better understand the memory initialization and testing.

enjoy-digital, 2022-04-04:
Hi @axibo-reiner,

I ordered some Tang Nano 9K but unfortunately haven't received yet them (and not sure I will unless I reorder some...), so I'm not able to test directly.

The PSRAM/HyperRAM support has been contributed by  @Icenowy and since this, the HyperRAM core has been reworked. I would first recommend testing with the HyperRAM core version from https://github.com/litex-hub/litehyperbus/releases/tag/2021.12. If working, then we have a regression on the HyperRAM core for Gowin parts and I could try to reorder some Tang Nano 9K, if failing, that's probably an issue with the toolchain.

axibo-reiner, 2022-04-16:
success

`
mCSR:		32-bit data
ROM:		64KiB
SRAM:		8KiB
FLASH:		4096KiB
MAIN-RAM:	4096KiB 

--========== Initialization ============--
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40000000 0B   
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Apr 16 2022 10:07:11
 BIOS CRC passed (aa7c93cc)

 LiteX git sha1: 3dbe349d

--=============== SoC ==================--
CPU:		VexRiscv_Debug @ 27MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		64KiB
SRAM:		8KiB
FLASH:		4096KiB
MAIN-RAM:	4096KiB 

--========== Initialization ============--
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 1.4MiB/s
   Read speed: 1.3MiB/s

Initializing W25Q32 SPI Flash @0x00000000...
SPI Flash clk configured to 13 MHz
Memspeed at 0 (Sequential, 4.0KiB)...
   Read speed: 1.4MiB/s
Memspeed at 0 (Random, 4.0KiB)...
   Read speed: 803.0KiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
No boot medium found

--============= Console ================--
`

Rolling back to 2021.12 for hyperbus worked with the latest tool chain from gowin. I will start poking around to see if i can upstream some fix.

enjoy-digital, 2022-06-03:
With https://github.com/litex-hub/litex-boards/commit/5188b17a7162b1cfe6b4e380d9cbd38e86b3a4dd, we are now using the old HyperRAM core on the tang nano 9k as a workaround until the issue is properly investigated.