litex-hub/litex-boards #382: Done DVI output on the Arty A7

suarezvictor, 2022-04-05:
It needs no special components, just wires!
[image]

Code snippet
```
class _CRG_arty(Module):
    def __init__(self, platform, sys_clk_freq, with_rst=True):
        self.rst = Signal()
        self.clock_domains.cd_sys       = ClockDomain()

        self.submodules.pll = pll = S7PLL(speedgrade=-1)
        rst    = ~platform.request("cpu_reset") if with_rst else 0
        self.comb += pll.reset.eq(rst | self.rst)
        pll.register_clkin(platform.request("clk100"), 100e6)
        pll.create_clkout(self.cd_sys,       sys_clk_freq)

        video_clock = 25e6 #"800x600@75Hz" =>  49.5e6, "640x480@60Hz" => 25.175e6 "800x600@60Hz"  => 40e6, 1280x720@60Hz(RB) => 61.9e6 1024 46.42e6
        if DVI:
            self.clock_domains.cd_hdmi   = ClockDomain()
            self.clock_domains.cd_hdmi5x = ClockDomain()
            pll.create_clkout(self.cd_hdmi,     video_clock, margin=1e-3)
            pll.create_clkout(self.cd_hdmi5x, 5*video_clock, margin=1e-3)
        else:
            #self.clock_domains.cd_vga       = ClockDomain(reset_less=True)
            self.clock_domains.cd_vga       = ClockDomain(reset_less=False) #TODO: chech why True brings errors
            pll.create_clkout(self.cd_vga, video_clock, margin=1e-3)
        platform.add_false_path_constraints(self.cd_sys.clk, pll.clkin) # Ignore sys_clk to pll.clkin path created by SoC's rst.

```
```

def build_arty(args):
	from litex_boards.platforms import arty as board
	platform = board.Platform(toolchain="vivado")
	#platform = board.Platform(toolchain="yosys+nextpnr") #brings errors about usage of DSP48E1 (* operator) and of ODDR
	sys_clk_freq = int(100e6)
	soc = SoCCore(platform, sys_clk_freq, **soc_core_argdict(args))
	soc.submodules.crg = _CRG_arty(platform, sys_clk_freq, False)
	soc.button = soc.platform.request('user_btn', 0)

	if DVI:
		from litex.soc.cores.video import VideoS7HDMIPHY
		platform.add_extension([("hdmi_out", 0, #DVI pmod breakout on pmod C (seems not working in others than C)
			Subsignal("data0_p", Pins("pmodc:0"), IOStandard("TMDS_33")),
			Subsignal("data0_n", Pins("pmodc:1"), IOStandard("TMDS_33")),
			Subsignal("data1_p", Pins("pmodc:2"), IOStandard("TMDS_33")),
			Subsignal("data1_n", Pins("pmodc:3"), IOStandard("TMDS_33")),
			Subsignal("data2_p", Pins("pmodc:4"), IOStandard("TMDS_33")),
			Subsignal("data2_n", Pins("pmodc:5"), IOStandard("TMDS_33")),
			Subsignal("clk_p",   Pins("pmodc:6"), IOStandard("TMDS_33")),
			Subsignal("clk_n",   Pins("pmodc:7"), IOStandard("TMDS_33")))])
		soc.submodules.videophy = VideoS7HDMIPHY(platform.request("hdmi_out"), clock_domain="hdmi")

```
[image]

enjoy-digital, 2022-04-21:
Great, thanks for sharing @suarezvictor! Just wondering, is your custom PMOD open-source? The Arty is a very nice and versatile development board and a DVI PMOD makes it even more interesting.

suarezvictor, 2022-04-21:
That PMOD is just one designed by someone else, see details here: https://hackaday.io/project/176327-hdmi-pmod

suarezvictor, 2022-04-21:
Same done with a female-female adapter, replacing one connector with the header
[image]

enjoy-digital, 2022-05-27:
Thanks @suarezvictor for sharing your great tests/findings. I'm closing to keep track of issues, but this will still be available for other users in closed issues.

closed 2022-05-27 by enjoy-digital