litex-hub/litex-boards #389: Misisng orangecrab -- or did I miss something?

sebeaumont, 2022-05-10:
```│ModuleNotFoundError: No module named 'litex_boards.platforms.orangecrab'```

sebeaumont, 2022-05-11:
Looks like its `gsd_orangecrab`

tcal-x, 2022-05-11:
Ah, yes there used to be short aliases provided so that you could use either `orangecrab` or `gsd_orangecrab`, but those were recently removed:

https://github.com/litex-hub/litex-boards/commit/683226df34d2a99aff1a190e6af9efabbb0282cd

I hadn't updated myself was was confused since I still saw `orangecrab` working for me.

enjoy-digital, 2022-05-27:
Hi @sebeaumont, @tcal-x,

short imports have indeed been removed (this was messing up python imports). Doing so has also be the occasion to detect/remove some inconsistencies, so I'm not necessarily thinking about adding it back (even if better implemented).
Sorry for the troubles.

closed 2022-05-27 by enjoy-digital