litex-hub/litex-boards #400: Support for Z7lite board

vanbwodonk, 2022-06-16:
Hi Dev,

I'm pretty new on Litex & Migen development. Trying to make support board Z7lite from Microphase (china). I buy it cheap from Aliexpress.
[image]

I already fork litex-board and make my own custom board.
https://github.com/vanbwodonk/litex-boards/blob/master/litex_boards/platforms/z7lite.py
https://github.com/vanbwodonk/litex-boards/blob/master/litex_boards/targets/z7lite.py
And this is board schematic:
https://github.com/vanbwodonk/zynq_z7lite_training/blob/main/Schematic/Z7-LITE_Rev1_1.pdf

Led charser already working with my z7lite targets .py. But i got error when activate the ethernet.
`python targets/z7lite.py --cpu-type None --with-ethernet --build`
Error:
```
 WARNING: [Synth 8-3936] Found unconnected internal register 'main_ethmac_tx_converter_converter_source_payload_data_reg' and it is trimmed from '10' to '9' bits. [/home/drw/LITEX/litex-boards/litex_boards/build/z7lite/gateware/z7lite.v:1752]
WARNING: [Synth 8-4767] Trying to implement RAM 'mem_3_reg' in registers. Block RAM or DRAM implementation is not possible; see log for reasons.
Reason is one or more of the following :
        1: RAM has multiple writes via different ports in same process. If RAM inferencing intended, write to one port per process. 
        2: Invalid write to RAM. 
        3: Unable to determine number of words or word size in RAM. 
        4: No valid read/write found for RAM.
```
Z7lite use RTL8201F phy. Please help.

enjoy-digital, 2022-06-23:
Hi @vanbwodonk,

sorry for the delay, I saw in your repo that you switched to RMII, has it fixed the issue? If not, I could have a look.

vanbwodonk, 2022-06-23:
Switch to MII because on board PHY (RTL8201F) is not gigabit. But still can't work. I need LAN for my project, so right now i used external module LAN8720 RMII with GPIO pins. The board has unconnected pin for CRS, COL, and RX_ER. Is that problem?