litex-hub/litex-boards #403: SoC option for cpu-count

smosanu, 2022-06-22:
In addition to the SoC option `--cpu-type`, for the cores already supported, it would be nice to enable the option `--cpu-count`, similar to the [linux-on-litex-vexriscv](https://github.com/litex-hub/linux-on-litex-vexriscv). Moreover, to test, it would be nice to add "corey?" (the core counter program) as a demo. Perhaps I can help with the development of this!? Ultimately, I would like to be able to generate SoCs with multiple naxriscv cores and run a benchmark that does take advantage of the cores.

enjoy-digital, 2022-06-23:
Hi @smosanu,

the CPU are now able to expose their own parameters and `--cpu-count `is exposed when selecting `--cpu-type=vexriscv_smp`:

`python3 -m litex_boards.targets.digilent_arty --cpu-type=vexriscv_smp --help`

will give you:

```
CPU options:
  --cpu-count CPU_COUNT
          Number of CPU(s) in the cluster. (default: 1)
  --with-coherent-dma
          Enable Coherent DMA Slave interface. (default: False)
  --without-coherent-dma
          Disable Coherent DMA Slave interface. (default: False)
  --dcache-width DCACHE_WIDTH
          L1 data cache bus width. (default: None)
  --icache-width ICACHE_WIDTH
          L1 instruction cache bus width. (default: None)
  --dcache-size DCACHE_SIZE
          L1 data cache size in byte per CPU. (default: None)
  --dcache-ways DCACHE_WAYS
          L1 data cache ways per CPU. (default: None)
  --icache-size ICACHE_SIZE
          L1 instruction cache size in byte per CPU. (default: None)
  --icache-ways ICACHE_WAYS
          L1 instruction cache ways per CPU (default: None)
  --aes-instruction AES_INSTRUCTION
          Enable AES instruction acceleration. (default: None)
  --without-out-of-order-decoder
          Reduce area at cost of peripheral access speed (default: False)
  --with-wishbone-memory
          Disable native LiteDRAM interface (default: False)
  --with-fpu
          Enable the F32/F64 FPU (default: False)
  --cpu-per-fpu CPU_PER_FPU
          Maximal ratio between CPU count and FPU count. Will instanciate as many FPU as necessary. (default: 4)
  --with-rvc
          Enable RISC-V compressed instruction support (default: False)
  --dtlb-size DTLB_SIZE
          Data TLB size. (default: 4)
  --itlb-size ITLB_SIZE
          Instruction TLB size. (default: 4)
```

For NaxRIscv, it's not yet SMP, so you'll have to do this manually, eventually with:
- https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_soc_gen.py
- https://github.com/enjoy-digital/litex_soc_gen_test/blob/master/digilent_arty.py

For the core counter program/benchmark, this can probably be developed externally to LiteX (at least for now).

smosanu, 2022-06-23:
Thank you, this is very helpful!