litex-hub/litex-boards #405: STLV7325 ethernet incorrect pin assignment

developandplay, 2022-06-27:
```
CRITICAL WARNING: [Vivado 12-1411] Cannot set LOC property of ports, Terminal eth0_col cannot be placed on W19 (IOB_X1Y7) because the pad is already occupied by terminal ddram_dqs_n[3] possibly due to user constraint [/home/martin/coding/linux-on-litex-vexriscv/build/aliexpress_stlv7325-100-eth/gateware/aliexpress_stlv7325.xdc:813]
CRITICAL WARNING: [Common 17-69] Command failed: 'R30' is not a valid site or package pin name. [/home/martin/coding/linux-on-litex-vexriscv/build/aliexpress_stlv7325-100-eth/gateware/aliexpress_stlv7325.xdc:817]
```
Unfortunately I couldn't find `col` or `crs` in the documentation for the STLV7325.
Is it fine to just leave them unconnected? [This](example/STLV7325/fpga_gmii/fpga.xdc) pull-request in Alex Forencich's ethernet module does exactly that. Not sure how important those two are in liteeth.

@enjoy-digital Did you ever get ethernet working on the STLV7325?
I couldn't with vexriscv-smp and the linux-on-litex-vexriscv has ethernet disabled by default.

enjoy-digital, 2022-07-05:
Hi @developandplay,

I'm don't think Ethernet has been tested on this hardware. `col `and `crs` pin should be not required by the LiteX build. I'll try to have a closer look when I'll have more time (and if you don't fix it before :)).

pftbest, 2022-09-30:
This two pins can be safely removed. They are not connected on the board. I think this is a typo in the platfrom file.

hansfbaier, 2023-04-13:
Those pins are not there anymore. I think this issue can be closed.

enjoy-digital, 2026-10-21:
The `col`/`crs` pins have been removed (and the platform is now `sitlinv_stlv7325_v1`/`sitlinv_stlv7325_v2`), so closing. Thanks!

closed 2026-10-21 by enjoy-digital