litex-hub/litex-boards #411: gw_sh can't find GoWin FPGA model

Chandler-Kluser, 2022-08-04:
tried to run:
```bash
python3 -m litex_boards.targets.sipeed_tang_nano_4k --build --load --cpu-variant=minimal
```
with litex + gcc-riscv + ninja + migen installed. I have compiled everything, but when the script runs `gw_sh` shell it fails in
```
set_device -name GW1NSR-4C GW1NSR-LV4CQN48PC7/I6
```
with the following error:
```
*** GOWIN Tcl Command Line Console  *** 
can't find device: GW1NSR-LV4CQN48PC7/I6 GW1NSR-4C

    while executing
"set_device -name GW1NSR-4C GW1NSR-LV4CQN48PC7/I6"
    (file "run.tcl" line 1)
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/root/litex/litex-boards/litex_boards/targets/sipeed_tang_nano_4k.py", line 169, in <module>
    main()
  File "/root/litex/litex-boards/litex_boards/targets/sipeed_tang_nano_4k.py", line 156, in main
    builder.build()
  File "/root/litex/litex/litex/soc/integration/builder.py", line 357, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/root/litex/litex/litex/soc/integration/soc.py", line 1277, in build
    return self.platform.build(self, *args, **kwargs)
  File "/root/litex/litex/litex/build/gowin/platform.py", line 41, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/root/litex/litex/litex/build/generic_toolchain.py", line 113, in build
    self.run_script(script)
  File "/root/litex/litex/litex/build/gowin/gowin.py", line 147, in run_script
    raise OSError("Error occured during Gowin's script execution.")
OSError: Error occured during Gowin's script execution.
```

I have used GoWin Education IDE versions 1.9.8.03 and 1.9.8 and both of them were not able to synthesize the sample project.

I have also tried to run `set_device -name GW1NSR-4C GW1NSR-LV4CQN48PC7/I6` in standalone `gw_sh` shell without the script, but I get the same error.

enjoy-digital, 2022-08-04:
Hi @Chandler-Kluser,

thanks for the feedback. The build has been tested with "GowinSynthesis V1.9.7.06Beta" and is still working with it. It seems Gowin changed things in  1.9.8, this would need to be investigated.

Chandler-Kluser, 2022-08-04:
Thanks for your quick reply, @enjoy-digital 

I can give more details if you request

enjoy-digital, 2022-08-04:
@Chandler-Kluser: I think we would just need to understand what's the expected device name for this chip with GoWin Education IDE 1.9.8.x

Chandler-Kluser, 2022-08-05:
The Tang Nano 4k chip part number is: `GW1NSR-LV4CQN48PC7/I6`

but giving this name it doesn't work, I have tried to run in a VM and in a docker container, but with no success, with the following IDE:

[image]

Running the `set_device` command from `gw_sh` I get:

[image]

It is important to mention that, in Sipeed's website and also GoWin's website, the Tang Nano 4K Part Number is:

[image]

Reading the manual page of the GW1NSR-4C devices, the part number syntax is:

[image]

And reading GoWin Software's Manual Documentation about the `set_device`command, I get:

[image]

but curiously, I get the same error reproducing the examples in `gw_sh`, I believe it is not a chip mispelling issue...

Can anyone help me to find out what is going on?

enjoy-digital, 2022-08-05:
@Chandler-Kluser: Could it be that this chip is not supported by the Education version? (I'm not aware of the limitation of the Education version).

Chandler-Kluser, 2022-08-05:
But it does support tang nano 4k! I have also tried another version and the same thing happened

Chandler-Kluser, 2022-08-05:
solved the issue!

GoWin Educational IDE accepts only `GW1NSR-LV4CQN48PC6/I5` part number, take a look at `gw_sh` output:
```
% set_device GW1NSR-LV4CQN48PC6/I5
current device: GW1NSR-4C  GW1NSR-LV4CQN48PC6/I5
```
I could check the Educational IDE by opening the wizard of the IDE for a new project:

[image]

I will open a pull request by updating `litex_boards/platforms/sipeed_tang_nano_4k.py` file to accept the Educational IDE.

closed 2022-08-05 by Chandler-Kluser