litex-hub/litex-boards #425: LimeSDR-mini-V2 Board ECP5 compiling issue and "Dirty" solution

Godtec, 2022-09-17:
Ran `./limesdr_mini_v2.py --csr-csv=csr.csv  --build` in directory: `litex-boards/litex_boards/targets`

Got Error:

.......  snip
> INFO:ECP5PLL:Creating ECP5PLL.
INFO:ECP5PLL:Registering Single Ended ClkIn of 40.00MHz.
INFO:ECP5PLL:Creating ClkOut0 sys of 80.00MHz (+-10000.00ppm).
Traceback (most recent call last):
  File "./limesdr_mini_v2.py", line 166, in <module>
    main()
  File "./limesdr_mini_v2.py", line 157, in main
    builder_kargs = trellis_argdict(args) if args.toolchain == "trellis" else {}
  File "/home/mikek/Documents/Cyclone5_SOC/Litex_New_release_Nov2021/litex/litex/build/lattice/trellis.py", line 260, in trellis_argdict
    "freq":         float(args.ecppack_freq),
TypeError: float() argument must be a string or a number, not 'NoneType'

Very "Dirty" Solution of commenting out line 260:

>  #"freq":         float(args.ecppack_freq),

Of file...

`litex/litex/build/lattice/trellis.py`

Then Compiled fine.

Thanks!
Mike Kennedy
VA3TEC

trabucayre, 2022-09-18:
I think you litex repo is not up to date: this failure was fixed by [this commit](https://github.com/enjoy-digital/litex/commit/ef58f681268c5c5777c5d266bc3de6875a22466e).

Gwen

Godtec, 2022-09-19:
Yes, Grovel... Sorry about that.

Godtec, 2022-09-19:
Closed, Ab_User error...  :(

closed 2022-09-19 by Godtec