litex-hub/litex-boards #434: can not build for artyz7 with cpu zynq7000

ruisukun21, 2022-10-13:
```bash
(litexenv) :~/litex_install/litex-boards$ python -m litex_boards.targets.digilent_arty_z7 --build --load
INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2022-10-13 23:44:15)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : xc7z020clg400-1.
INFO:SoC:System clock: 125.000MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Controller ctrl added.
INFO:SoC:CPU zynq7000 added.
INFO:SoC:CPU zynq7000 adding IO Region 0 at 0x40000000 (Size: 0xbc000000).
INFO:SoCRegion:Region size rounded internally from 0xbc000000 to 0x100000000.
INFO:SoCBusHandler:io0 Region added at Origin: 0x40000000, Size: 0xbc000000, Mode: RW, Cached: False Linker: False.
INFO:SoC:CPU zynq7000 setting reset address to 0xfc000000.
INFO:SoC:CPU zynq7000 adding Bus Master(s).
--2022-10-13 23:44:15--  http://kmf2.trabucayre.com/arty_z7_20.tcl
Resolving kmf2.trabucayre.com (kmf2.trabucayre.com)... 91.121.155.27
Connecting to kmf2.trabucayre.com (kmf2.trabucayre.com)|91.121.155.27|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 31833 (31K) [text/x-tcl]
Saving to: ‘arty_z7_20.tcl.1’

arty_z7_20.tcl.1                             100%[=============================================================================================>]  31.09K   125KB/s    in 0.2s    

2022-10-13 23:44:16 (125 KB/s) - ‘arty_z7_20.tcl.1’ saved [31833/31833]

INFO:SoCBusHandler:master0 added as Bus Master.
INFO:SoC:CSR Bridge csr added.
INFO:SoCBusHandler:csr Region added at Origin: 0x40000000, Size: 0x00010000, Mode: RW, Cached: False Linker: False.
INFO:SoCBusHandler:csr added as Bus Slave.
INFO:SoCCSRHandler:csr added as CSR Master.
INFO:SoCBusHandler:Interconnect: InterconnectShared (1 <-> 1).
INFO:SoCCSRHandler:ctrl CSR allocated at Location 0.
INFO:SoCCSRHandler:identifier_mem CSR allocated at Location 1.
INFO:SoCCSRHandler:leds CSR allocated at Location 2.
INFO:SoCCSRHandler:timer0 CSR allocated at Location 3.
ERROR:SoC:CPU needs reset address 0xfc000000 to be in a defined Region.
ERROR:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
IO Regions: (1)
io0                 : Origin: 0x40000000, Size: 0xbc000000, Mode: RW, Cached: False Linker: False
Bus Regions: (1)
csr                 : Origin: 0x40000000, Size: 0x00010000, Mode: RW, Cached: False Linker: False
Bus Masters: (1)
- master0
Bus Slaves: (1)
- csr
```

related issue #358

ruisukun21, 2022-10-13:
another related issue #423

ruisukun21, 2022-10-13:
I tried the solution in https://github.com/litex-hub/litex-boards/pull/423#issuecomment-1245813656, but now I get 100 critical warning and led chaser is not working. Note: I get led chaser when I use ` --cpu-type=vexriscv`.
```
CRITICAL WARNING: [Common 17-69] Command failed: 'X' is not a valid site or package pin name. [~/litex_install/litex-boards/build/digilent_arty_z7/gateware/digilent_arty_z7.xdc:5]
``` 
[critical_warning.txt](https://github.com/litex-hub/litex-boards/files/9779046/critical_warning.txt)

trabucayre, 2022-10-13:
Critical warning may be neglected: it's okay it may be required to modify xdc filling to simply avoid to add lines related to **PS** part.
I have opened a PR (#435) to add flash region (and to fix this issue).

For led chaser:
Your board is configured (JP4) to boot on JTAG? I have the same issue because the CPU is not configured -> clock from PS may not configured. When I swith to SDCard with a FSBL it's working.

The BIOS part has to be added. I will do that ASAP.

ruisukun21, 2022-10-17:
Thank you very much for your reply.
>>Your board is configured (JP4) to boot on JTAG?
yes.
>>When I swith to SDCard with a FSBL it's working.
I see. I am new to litex. It is not clear to me how to prepare the SDcard with a FSBL .
could you point me in the right direction?

Allow me to ask here something a little bit off topic.
I wonder why `no_uart` flag/argument is forced to true with set_defaults (as its action field is store_true, it should be False by default):
`parser.set_defaults(no_uart=True) ` 
(https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/digilent_arty_z7.py#L110, https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/soc_core.py#L344)
Also wonder, why uart is forced to False for Zynq cpus `kwargs["with_uart"]            = False`
(https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/digilent_arty_z7.py#L65).
 Is there a limitation that must be resolved for using uart?

trabucayre, 2022-10-18:
> Thank you very much for your reply.
> 
> > > Your board is configured (JP4) to boot on JTAG?
> 
> yes.
> 
> > > When I swith to SDCard with a FSBL it's working.
> 
> I see. I am new to litex. It is not clear to me how to prepare the SDcard with a FSBL . could you point me in the right direction?
> 

For this type of board/device I, usually, build a system using buildroot. Currently arty z7 is not officially supported but you could use this [repository](https://github.com/trabucayre/buildroot-external). You have to do:

```
git clone https://git.busybox.net/buildroot
git clone https://github.com/trabucayre/buildroot-external.git
source buildroot-external/sourceme.ggm
cd buildroot
make digilent_arty_z7_defconfig
make
```
system image is located under **output/images/** and called **sdcard.img**

To flash your sd-card:
```
sudo dd if=output/images/sdcard.img of=/dev/sdXX bs=4M
```
Keep in mind to double (triple) check your **/dev/sdxxx** to avoid to destroy your hard drive content.

> Allow me to ask here something a little bit off topic. I wonder why `no_uart` flag/argument is forced to true with set_defaults (as its action field is store_true, it should be False by default): [...]

UART interface is connected to PS (CPU) side. If you want to use a softcore you have to connect an external USB<->UART device.

ruisukun21, 2022-11-14:
It worked!! 
@trabucayre  Thank you very much for your help.
The first time I attempt your instructions, I got a problem with linux sources and I solved it following the fix of this link:
https://lore.kernel.org/buildroot/20221004085922.3c755666@gmx.net/
But after that fix, I got a problem related to vivado. 
Today (it's been a while), I try again. The repository was updated at the beginning of the make script and it ran until the end. 
I flashed the sdcard and linux started working after resetting the FPGA board !
Thank you so much!

trabucayre, 2022-11-14:
Happy this could help.
If my **buildroot-external** repo needs some fix don't hesitate to open an issue or a PR :)
It's interesting I haven't seen this issue (but I usually uses buildroot master instead of a release).

I have also updated this board: now it's possible to use LiteX bios instead of Linux.