litex-hub/litex-boards #439: Colorlight 75E hardware 8.0 is not yet done

cdwijs, 2022-10-22:
I'm the proud owner of a Colorlight 75E hardware version 8.0. The connections from the FPGA to the connectors are not yet mapped.

I would like to create gateware that shows what ball of the FPGA goes to what point on the headers.

On each pin, I would like to output a train of pulses, that indicate the ball. First a preamble, and then 2 burst indicating the row and column That way I can look at an LED to figure out the connection.

For example:
A2: 10101010101010000010000010010000 (and then repeat)
G4: 10101010101010000010010010010010010010000010010010010000
C4: 10101010101010000010010010000010010010010000

Has this already been done?
Where can I find a file that defines the balls of the FPGA package?

cdwijs, 2022-10-22:
Here's the mapping between symbols and balls of the FPGA:

https://www.latticesemi.com/view_document?document_id=50485

enjoy-digital, 2022-10-22:
Hi @cdwijs,

you can find here a LiteX design that I used to document one of the variant: https://github.com/enjoy-digital/colorlite/blob/master/ios_stream.py

This should allow you to create your own version for the 8.0 variant.

rowanG077, 2022-10-23:
There is an LPF file here: https://github.com/q3k/chubby75/blob/master/5a-75b/lpf/cl-5a-75v-v80.lpf

But I didn't test it so I can't vouch for the validity. But at least it might be easiest to just copy over that first and with any luck it will work!

enjoy-digital, 2026-10-21:
Hi @cdwijs, sorry for the very long delay. Some updates since this was opened:

- The LPF linked above (`cl-5a-75v-v80.lpf`) is for the **5A-75B** v8.0, which is supported with `colorlight_5a_75b --revision 8.0`.
- For the **5A-75E**, a v8.2 platform has been added (`colorlight_5a_75e --revision 8.2`, LFE5U-25F-7BG256I). The v8.0 is not documented anywhere that we know of, but it is likely close to the v8.2, so this is probably the best starting point:
  ```
  python3 -m litex_boards.targets.colorlight_5a_75x --board 5a-75e --revision 8.2 --build --load
  ```

Could you try it on your board and tell us if it works (serial/LEDs, and Ethernet if possible)? If it does, we can just accept `8.0` as an alias of `8.2`. If some pins differ, the `ios_stream.py` approach from colorlite mentioned above can be used to find them.