litex-hub/litex-boards #444: Fireant build error "spiflash Region in IO region, it can't be cached"

f-yildirim, 2022-10-28:
I've just pulled the master branch and trying to build vexriscv for the target jungle_electronics_fireant.
Without any modifications it fails with error "spiflash Region in IO region, it can't be cached"
Modifying the spi flash address 0x70000000 build is successful, but this time no response from with serial with the parameters below. Double checked that I've been using the right gpio pins on the board (GPIOR_31 & GPIOR_32).

```
# Default peripherals
serial = [
    ("serial", 0,
        Subsignal("tx", Pins("J2:2")),
        Subsignal("rx", Pins("J2:3")),
        IOStandard("3.3_V_LVTTL_/_LVCMOS") , Misc("WEAK_PULLUP")
    )
]

_connectors = [
    ["J1", " - G5 G4 J3 G3 J2 H2 F3 G1 F1 E2 E1 C2 D2 E3 D3 B3 A5 B5 -"],
    ["J2", " -  - H8 G8 H9 G9 F5 F6 F7 E7 F8 E8 D9 B9 C8 B8 A8 C6 A6 -"],
]
```

```
$ ./jungle_electronics_fireant.py --build
block pll: names:['PLL_0']
INFO:EFINIXPLL:Creating TRIONPLL
INFO:EFINIXPLL:Clock source: EXTERNAL, using EXT_CLK3
INFO:EFINIXPLL:PLL used     : ['PLL_0']
INFO:EFINIXPLL:PLL available: []
INFO:EFINIXPLL:Use PLL_0
INFO:EFINIXPLL:Creating ClkOut0 pll0_clkout0 of 33.33MHz (+-0.00ppm).
INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2022-10-28 13:20:46)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : T8F81.
INFO:SoC:System clock: 33.333MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Controller ctrl added.
INFO:SoC:CPU vexriscv added.
INFO:SoC:CPU vexriscv adding IO Region 0 at 0x80000000 (Size: 0x80000000).
INFO:SoCBusHandler:io0 Region added at Origin: 0x80000000, Size: 0x80000000, Mode: RW, Cached: False Linker: False.
INFO:SoC:CPU vexriscv overriding sram mapping from 0x01000000 to 0x10000000.
INFO:SoC:CPU vexriscv setting reset address to 0x80040000.
INFO:SoC:CPU vexriscv adding Bus Master(s).
INFO:SoCBusHandler:cpu_bus0 added as Bus Master.
INFO:SoCBusHandler:cpu_bus1 added as Bus Master.
INFO:SoC:CPU vexriscv adding Interrupt(s).
INFO:SoC:CPU vexriscv adding SoC components.
INFO:SoCBusHandler:sram Region added at Origin: 0x10000000, Size: 0x00002000, Mode: RWX, Cached: True Linker: False.
INFO:SoCBusHandler:sram added as Bus Slave.
INFO:SoC:RAM sram added Origin: 0x10000000, Size: 0x00002000, Mode: RWX, Cached: True Linker: False.
INFO:SoCIRQHandler:uart IRQ allocated at Location 0.
INFO:SoCIRQHandler:timer0 IRQ allocated at Location 1.
ERROR:SoCBusHandler:spiflash Region in IO region, it can't be cached: Origin: 0x80000000, Size: 0x00100000, Mode: RW, Cached: True Linker: False
ERROR:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
IO Regions: (1)
io0                 : Origin: 0x80000000, Size: 0x80000000, Mode: RW, Cached: False Linker: False
Bus Regions: (1)
sram                : Origin: 0x10000000, Size: 0x00002000, Mode: RWX, Cached: True Linker: False
Bus Masters: (2)
- cpu_bus0
- cpu_bus1
Bus Slaves: (1)
- sram
```

enjoy-digital, 2022-12-08:
Thanks @f-yildirim for reporting, the design has been updated with https://github.com/litex-hub/litex-boards/commit/12db52471d0a9b154566f91f47ef81b9c03bb1e5 and is now similar to other boards executing the BIOS directly from SPIFlash.

enjoy-digital, 2022-12-08:
I'm closing since this now should work but please re-open of this is not the case.

closed 2022-12-08 by enjoy-digital

f-yildirim, 2022-12-08:
Thanks @enjoy-digital for you effort. Build is successful now.
But, it seems that there is still an issue with serial rx/tx. I will double check and open another issue if needed.
Thanks..