litex-hub/litex-boards #446: Wrong pin for HDMI in Decklink Mini 4K ?

rdolbeau, 2022-11-06:
Hello,

Looking at this:

https://github.com/litex-hub/litex-boards/blob/61eb39101af6b83be1ee84b03a07769dc6344bc5/litex_boards/platforms/decklink_mini_4k.py#L113-L114

It doesn't make sense to me ; from [https://www.xilinx.com/content/dam/xilinx/support/packagefiles/a7packages/xc7a100tfgg676pkg.txt](https://www.xilinx.com/content/dam/xilinx/support/packagefiles/a7packages/xc7a100tfgg676pkg.txt)  

AE9   MGTPTXP1_213                  NA                 213   NA            NA                  GTP       NA
(...)
AD10  MGTPTXN0_213                  NA                 213   NA            NA                  GTP       NA

that's not a pair. Also AD10 is used in https://github.com/litex-hub/litex-boards/blob/61eb39101af6b83be1ee84b03a07769dc6344bc5/litex_boards/platforms/decklink_mini_4k.py#L100, where it makes sense.

I think the AD10 should actually be AF9:
AF9   MGTPTXN1_213                  NA                 213   NA            NA                  GTP       NA

Cordially,

rdolbeau, 2022-11-08:
@enjoy-digital Sorry I don't have the board to test, just looking at how HDMI was implemented using GTPs.

enjoy-digital, 2022-11-08:
@rdolbeau: Ah OK, I'll do some implementation tests to confirm this then.

enjoy-digital, 2022-11-14:
@rdolbeau: I just had a closer look. Vivado was indeed automatically re-assigning the pin to AF9:

enjoy-digital, 2022-11-14:
Fixed with https://github.com/litex-hub/litex-boards/commit/6e10df234fbf87b3fffb130d4fc7c0251fcf84da.

closed 2022-11-14 by enjoy-digital