litex-hub/litex-boards #454: icebreaker board:  Failed to expand region (0, 0) |_> (25, 31) of 5491 ICESTORM_LCs

disdi, 2022-11-19:
I am facing issue with building for Icebraker board:
`python3 -m litex_boards.targets.icebreaker --build`

[image]

Yosys version I am using  :
`Yosys 0.22+82 (git sha1 2cdbb85da, clang 10.0.0-4ubuntu1 -fPIC -Os)`

disdi, 2022-11-20:
Tried to build Yosys and nextpnr from source to get latest version but still getting same error as shown below : 

[image]

tcal-x, 2022-11-21:
Hi @disdi , the error message is a bit cryptic, but it's just saying that your mapped design uses more resources than are available.   In this case, you've used 5481 ICESTORM_LCs while the target FPGA has only 5280.

The default CPU variant does not fit on the target FPGA.   While you might wonder why they don't make the default to be something that fits for this board, I think the reasoning is that it would cause confusion if every board had a different default.   So globally, the default cpu type is VexRiscv, and the default variant is "full".

These are some recipes that should work, using smaller CPUs:

VexRiscv, "minimal":
```
python3 -m litex_boards.targets.icebreaker --build --cpu-variant=minimal
```
or VexRiscv, "lite":
```
python3 -m litex_boards.targets.icebreaker --build --cpu-variant=lite
```
or bit-serial SERV:
```
python3 -m litex_boards.targets.icebreaker --build --cpu-type=serv
```

enjoy-digital, 2022-11-22:
Thanks @tcal-x, we should maybe make sure to provide a default configuration that compiles fine on all boards. I'll think about the best way to do it.

trabucayre, 2022-11-22:
@enjoy-digital maybe doing something like done for [tang nano 4k](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/sipeed_tang_nano_4k.py#L85): but instead of raising an exception, print a message and forcing `minimal` ?

enjoy-digital, 2022-11-22:
@trabucayre: Yes, different approaches have probably been used, it would be good to try to do something similar for all boards (and with minimal changes). Using `parser.set_defaults(cpu_variant="lite")` could also be interesting and minimize the changes.

trabucayre, 2022-11-22:
LGTM but in `parser.py` cpu`s args are injected in `parse_args` and consequently `set_default` is lost.
I suppose overriding `set_defaults` to intercept some params like this is required to reinject them just before calling `argparse.ArgumentParser.parse_args`

maverickcasanova, 2022-11-23:
I confirm the  --cpu-variant=lite and --cpu-variant=minimal both work for me.
Let me know if I need to close this bug or do we expect a pull request to change the default option here ?

disdi, 2022-11-23:
I confirm the --cpu-variant=lite and --cpu-variant=minimal both work for me.
Let me know if I need to close this bug or do we expect a pull request to change the default option here ?

disdi, 2022-11-24:
Closing this issue for now

closed 2022-11-24 by disdi