litex-hub/litex-boards #466: digilent_arty fails to build using f4pga due to dependency on unsupported features (DNA_PORT, XADC)

mlaga97, 2023-01-06:
Building digilent_arty.py with `--toolchain vivado` succeeds, but when built with `--toolchain f4pga` the build fails with the following error:

```
Found constant-zero generator '$abc$82080$auto$blifparse.cc:518:parse_blif$89937.T0'
# Load circuit took 0.35 seconds (max_rss 90.7 MiB, delta_rss +32.2 MiB)
The entire flow of VPR took 1.21 seconds (max_rss 90.7 MiB)

stderr:
Error 1: 
Type: Blif file
File: /home/osboxes/src/synth/build/digilent_arty/gateware/digilent_arty.eblif
Line: 184330
Message: Failed to find matching architecture model for 'DNA_PORT'
```

Per the https://github.com/chipsalliance/f4pga/issues/648 thread, this issue occurs because the DNA_PORT block is unsupported by f4pga.

Commenting out https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/digilent_arty.py#L97 results in a similar error about the 'XADC' block, which is also unsupported by f4pga.

Once both the call to `DNA()` and `XADC()` are commented out, then a bitstream can be successfully generated with the f4pga toolchain and is able to be booted on the device itself.

trabucayre, 2023-01-06:
Yes! It's true. `XADC` and `DNA` must be added conditionnaly only when `toolchain == "Vivado"`
Make sense to add a test for arty target (and all others with this features)

mithro, 2023-01-07:
Please make sure that you log issues on the F4PGA tools about missing support for these primitives? Then you can put a comment in LiteX to remove the disable once the bug is closed because the primitive has been added.

mithro, 2023-01-07:
FYI - @kgugala

mlaga97, 2023-01-07:
> Please make sure that you log issues on the F4PGA tools about missing support for these primitives? Then you can put a comment in LiteX to remove the disable once the bug is closed because the primitive has been added.

I searched, and there appears to be a closed issue on that exact topic here:
https://github.com/chipsalliance/f4pga/issues/623

Additionally, there is a comment in the chipsalliance/f4pga#648 thread mentioning investigating the difficulty of adding DNA_PORT support.

PR #467 (2023-01-07): Allow building digilent_arty using f4pga
Only use XADC() and DNA() functions if vivado is the current toolchain.

I believe that this fixes #466 and https://github.com/chipsalliance/f4pga/issues/648
closed 2023-01-07 by enjoy-digital