litex-hub/litex-boards #481: Sipeed Tang Nano 9k no Boot Message from SoC-Core

maggo1404, 2023-03-10:
Hi,

i Build with the Standard config the bitfile. 
`sipeed_tang_nano_9k.py —build` 

When finish load to the SRAM or Flash the bitfile.

The LEDs performe a running light Show. 
Open the serial Port with minicom (115200 8N1) and push the reset Button i hav never sean the boot Message from bios. 

When the build with `litex_sim —build` then see the boot Message. 

OS Lububtu 22.04 LTS

Regards maggo1404

trabucayre, 2023-03-10:
Not fully sure but [rom is in spiflash](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/sipeed_tang_nano_9k.py#L85)
Could you try with:
```
./sipeed_tang_nano_9k.py --prog-kit openfpgaloader --flash --build
```

maggo1404, 2023-03-12:
is not sucessfull :(

i have the small idea the VM is not the perfect Environment for the programming a FPGA-board. 
[image]

have you the settings for the gowin programmer? I can try with the windows-versionen. I have search and havn't found the perfect setting. :(

maggo1404, 2023-03-12:
reboot the FPGA-Board and PC and now running the soft-Core. 
[image]

The writing for the bitfile is very slow. Thanks for the support.

closed 2023-03-12 by maggo1404

trabucayre, 2023-03-12:
JTAG interface is quite unstable in some case :-(
Never tested with a VM nor with windows.
For time to write: SPI flash is accessed with a bitbanging mode and can't be really fast.

Great to see everything is now working!