litex-hub/litex-boards #496: VCU128 DDR4 memory calibration failure

jiegec, 2023-04-19:
Hi, I am porting litex-boards to VCU128, and I encountered DDR4 memory calibration failure:

```shell
--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 556
  Cmd/Clk scan (0-278)
  |00011  |111111111  |111111111  |111111111| best: 340
  Setting Cmd/Clk delay to 340 taps.
  Data scan:
  m0: |111111111111111111111111| delay: 00
  m1: |011111111111111111100000| delay: 11
  m2: |111111111111111111111111| delay: 00
  m3: |000111111111111111110000| delay: 37
  m4: |111111111111111111111111| delay: 00
  m5: |000000011111111111111111| delay: 112
  m6: |000011111111111111111100| delay: 58
  m7: |111111111111111111111111| delay: 00
Write latency calibration:
m0:2 m1:6 m2:2 m3:6 m4:2 m5:6 m6:6 m7:2
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b07 delays: -
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |11111100000000000000000000000000| delays: 41+-41
  m1, b04: |00000000001111111111111000000000| delays: 252+-107
  m1, b05: |00000000000000000000000000011111| delays: 470+-41
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b04 delays: 252+-107
  m2, b00: |00000000000000000000000000000000| delays: -
  m2, b01: |00000000000000000000000000000000| delays: -
  m2, b02: |00000000000000000000000000000000| delays: -
  m2, b03: |00000000000000000000000000000000| delays: -
  m2, b04: |00000000000000000000000000000000| delays: -
  m2, b05: |00000000000000000000000000000000| delays: -
  m2, b06: |00000000000000000000000000000000| delays: -
  m2, b07: |00000000000000000000000000000000| delays: -
  best: m2, b07 delays: -
  m3, b00: |00000000000000000000000000000000| delays: -
  m3, b01: |00000000000000000000000000000000| delays: -
  m3, b02: |00000000000000000000000000000000| delays: -
  m3, b03: |00000000000000000000000000000000| delays: -
  m3, b04: |00111111111111111000000000000000| delays: 148+-116
  m3, b05: |00000000000000000000111111111111| delays: 413+-98
  m3, b06: |00000000000000000000000000000000| delays: -
  m3, b07: |00000000000000000000000000000000| delays: -
  best: m3, b04 delays: 146+-115
  m4, b00: |00000000000000000000000000000000| delays: -
  m4, b01: |00000000000000000000000000000000| delays: -
  m4, b02: |00000000000000000000000000000000| delays: -
  m4, b03: |00000000000000000000000000000000| delays: -
  m4, b04: |00000000000000000000000000000000| delays: -
  m4, b05: |00000000000000000000000000000000| delays: -
  m4, b06: |00000000000000000000000000000000| delays: -
  m4, b07: |00000000000000000000000000000000| delays: -
  best: m4, b07 delays: -
  m5, b00: |00000000000000000000000000000000| delays: -
  m5, b01: |00000000000000000000000000000000| delays: -
  m5, b02: |00000000000000000000000000000000| delays: -
  m5, b03: |00000000000000000000000000000000| delays: -
  m5, b04: |11111111111100000000000000000000| delays: 90+-90
  m5, b05: |00000000000000011111111111111000| delays: 349+-111
  m5, b06: |00000000000000000000000000000000| delays: -
  m5, b07: |00000000000000000000000000000000| delays: -
  best: m5, b05 delays: 350+-111
  m6, b00: |00000000000000000000000000000000| delays: -
  m6, b01: |00000000000000000000000000000000| delays: -
  m6, b02: |00000000000000000000000000000000| delays: -
  m6, b03: |00000000000000000000000000000000| delays: -
  m6, b04: |00001111111111111000000000000000| delays: 166+-103
  m6, b05: |00000000000000000000001111111111| delays: 427+-83
  m6, b06: |00000000000000000000000000000000| delays: -
  m6, b07: |00000000000000000000000000000000| delays: -
  best: m6, b04 delays: 168+-106
  m7, b00: |00000000000000000000000000000000| delays: -
  m7, b01: |00000000000000000000000000000000| delays: -
  m7, b02: |00000000000000000000000000000000| delays: -
  m7, b03: |00000000000000000000000000000000| delays: -
  m7, b04: |00000000000000000000000000000000| delays: -
  m7, b05: |00000000000000000000000000000000| delays: -
  m7, b06: |00000000000000000000000000000000| delays: -
  m7, b07: |00000000000000000000000000000000| delays: -
  best: m7, b07 delays: -
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
  bus errors:  256/256
  addr errors: 0/8192
  data errors: 524277/524288
Memtest KO
Memory initialization failed
```

It is possible to workaround the issue by enabling only 1,3,5,6 modules:

```python
                # pads             = platform.request("ddram"),
                pads             = PHYPadsReducer(platform.request("ddram"), [1, 3, 5, 6]),
```

But I would like the investigate the issue of the other four.

My code can be found at https://github.com/jiegec/litex-boards/commit/d71c29055a3343e0194bafa47a05434660cc9cd2

What I have tried:

- change idelay frequency
- set cmd_latency to 1
- change sys clock frequency
- uncomment ten/alert_n/par signals
- use sdram_force_cmd_delay

Calibration result of Xillinx MIG IP:

[image]

jiegec, 2023-04-19:
I isolated the module 0 via:

```python
                pads             = PHYPadsReducer(platform.request("ddram"), [0]),
```

The calibration result:

```
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 560
  Cmd/Clk scan (0-280)
  |11111  |11111  |11111  |11111| best: 0
  Setting Cmd/Clk delay to 0 taps.
  Data scan:
  m0: |111111111111111111111111| delay: 00
Write latency calibration:
m0:2
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b00 delays: -
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
  bus errors:  136/256
  addr errors: 0/8192
  data errors: 524288/524288
Memtest KO
```

jiegec, 2023-04-19:
By reading the datasheet, module 0, 2, 4, 7 corresponds to U74 and U73. They are located in the bottom side of clam-shell.

jiegec, 2023-04-19:
After more investigation, address mirroring is applied to bottom DRAMs: <https://support.xilinx.com/s/question/0D52E00006tceu6SAA/ultrascalempsoc-pl-ddr4-clamshell-address-mirroring?language=en_US>

jiegec, 2023-04-19:
I manually swapped the address lines:

```python
        Subsignal("a", Pins(
            # "BF50 BD51 BG48 BE50 BE49 BE51 BF53 BG50",
            # "BF51 BG47 BF47 BG49 BF48 BF52"),
            # Swap A3A4, A5A6, A7A8, A11A13
            "BF50 BD51 BG48 BE49 BE50 BF53 BE51 BF51",
            "BG50 BG47 BF47 BF52 BF48 BG49"),
            IOStandard("SSTL12_DCI")),
        # Subsignal("ba",        Pins("BE54 BE53"), IOStandard("SSTL12_DCI")),
        # Swap BA0/1
        Subsignal("ba",        Pins("BE53 BE54"), IOStandard("SSTL12_DCI")),
        Subsignal("cs_n",  Pins("BK48"), IOStandard("SSTL12_DCI")), # Clam-shell fashion
        #Subsignal("cs_n",      Pins("BP49"), IOStandard("SSTL12_DCI")),
        #Subsignal("bot_cs_n",  Pins("BK48"), IOStandard("SSTL12_DCI")), # Clam-shell fashion
```

In this way, the bottom DRAMs can be calibrated successfully. However, the top and bottom ones can not be used at the same time.

jiegec, 2023-04-19:
To sum up, litedram need to be extended to support:

1. Mirrored Pins: A3A4, A5A6, A7A8, A11A13, BA0BA1, BG0BG1
2. Clam shell fashion: like two ranks but with private data pins

jiegec, 2023-04-20:
litedram already considers address mirroring for RDIMMs:

```python
            if phy_settings.is_rdimm:
                assert phy_settings.memtype == "DDR4"
                # JESD82-31A page 38
                #
                # B-side chips have certain usually-inconsequential address and BA
                # bits inverted by the RCD to reduce SSO current. For mode register
                # writes, however, we must compensate for this. BG[1] also directs
                # writes either to the A side (BG[1]=0) or B side (BG[1]=1)
                #
                # The 'ba != 7' is because we don't do this to writes to the RCD
                # itself.
                if ba != 7:
                    invert_masks.append((0b10101111111000, 0b1111))
```

jiegec, 2023-04-20:
The DFII interface replicates the cs_n bits:

```python
                phase.cs_n.eq(Replicate(~self._command.fields.cs, len(phase.cs_n))),
                phase.we_n.eq(~self._command.fields.we),
                phase.cas_n.eq(~self._command.fields.cas),
                phase.ras_n.eq(~self._command.fields.ras)
```

We need to expose the two cs_n bits to software upon calibration. In later places, the two cs_n can be used as one.

jiegec, 2023-04-20:
I am working on a dirty fix:

- Change csr_dfi in DFIInjector to allow software to control top/bottom cs_n separately
- Swap address pins in software and send the same commands to top/bottom chip separately

jiegec, 2023-04-20:
I made it!

```
--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 556
  Cmd/Clk scan (0-278)
  |00011  |111111111  |111111111  |111111111| best: 334
  Setting Cmd/Clk delay to 334 taps.
  Data scan:
  m0: |000111111111111111111000| delay: 37
  m1: |011111111111111111100000| delay: 05
  m2: |001111111111111111110000| delay: 26
  m3: |001111111111111111110000| delay: 31
  m4: |000011111111111111111100| delay: 64
  m5: |000000011111111111111111| delay: 106
  m6: |000011111111111111111000| delay: 53
  m7: |000011111111111111111000| delay: 54
Write latency calibration:
m0:6 m1:6 m2:6 m3:6 m4:6 m5:6 m6:6 m7:6
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |11100000000000000000000000000000| delays: 22+-22
  m0, b04: |00000000111111111111100000000000| delays: 217+-102
  m0, b05: |00000000000000000000000001111111| delays: 454+-56
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b04 delays: 215+-104
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |11111100000000000000000000000000| delays: 42+-42
  m1, b04: |00000000001111111111111000000000| delays: 255+-103
  m1, b05: |00000000000000000000000000001111| delays: 475+-35
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b04 delays: 255+-103
  m2, b00: |00000000000000000000000000000000| delays: -
  m2, b01: |00000000000000000000000000000000| delays: -
  m2, b02: |00000000000000000000000000000000| delays: -
  m2, b03: |11100000000000000000000000000000| delays: 15+-15
  m2, b04: |00000001111111111111000000000000| delays: 203+-105
  m2, b05: |00000000000000000000000001111111| delays: 447+-64
  m2, b06: |00000000000000000000000000000000| delays: -
  m2, b07: |00000000000000000000000000000000| delays: -
  best: m2, b04 delays: 203+-107
  m3, b00: |00000000000000000000000000000000| delays: -
  m3, b01: |00000000000000000000000000000000| delays: -
  m3, b02: |00000000000000000000000000000000| delays: -
  m3, b03: |00000000000000000000000000000000| delays: -
  m3, b04: |00011111111111111000000000000000| delays: 147+-108
  m3, b05: |00000000000000000000011111111111| delays: 416+-95
  m3, b06: |00000000000000000000000000000000| delays: -
  m3, b07: |00000000000000000000000000000000| delays: -
  best: m3, b04 delays: 149+-111
  m4, b00: |00000000000000000000000000000000| delays: -
  m4, b01: |00000000000000000000000000000000| delays: -
  m4, b02: |00000000000000000000000000000000| delays: -
  m4, b03: |00000000000000000000000000000000| delays: -
  m4, b04: |11111111111110000000000000000000| delays: 102+-102
  m4, b05: |00000000000000000111111111111100| delays: 373+-105
  m4, b06: |00000000000000000000000000000000| delays: -
  m4, b07: |00000000000000000000000000000000| delays: -
  best: m4, b05 delays: 375+-104
  m5, b00: |00000000000000000000000000000000| delays: -
  m5, b01: |00000000000000000000000000000000| delays: -
  m5, b02: |00000000000000000000000000000000| delays: -
  m5, b03: |00000000000000000000000000000000| delays: -
  m5, b04: |11111111111000000000000000000000| delays: 89+-89
  m5, b05: |00000000000000001111111111111000| delays: 354+-104
  m5, b06: |00000000000000000000000000000000| delays: -
  m5, b07: |00000000000000000000000000000000| delays: -
  best: m5, b05 delays: 354+-106
  m6, b00: |00000000000000000000000000000000| delays: -
  m6, b01: |00000000000000000000000000000000| delays: -
  m6, b02: |00000000000000000000000000000000| delays: -
  m6, b03: |00000000000000000000000000000000| delays: -
  m6, b04: |00000111111111111000000000000000| delays: 169+-97
  m6, b05: |00000000000000000000000111111111| delays: 432+-79
  m6, b06: |00000000000000000000000000000000| delays: -
  m6, b07: |00000000000000000000000000000000| delays: -
  best: m6, b04 delays: 172+-100
  m7, b00: |00000000000000000000000000000000| delays: -
  m7, b01: |00000000000000000000000000000000| delays: -
  m7, b02: |00000000000000000000000000000000| delays: -
  m7, b03: |00000000000000000000000000000000| delays: -
  m7, b04: |11111111111100000000000000000000| delays: 93+-93
  m7, b05: |00000000000000000111111111111000| delays: 365+-102
  m7, b06: |00000000000000000000000000000000| delays: -
  m7, b07: |00000000000000000000000000000000| delays: -
  best: m7, b05 delays: 363+-101
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 108.8MiB/s
   Read speed: 93.6MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
No boot medium found

--============= Console ================--

litex> 
```

jiegec, 2023-04-20:
You can find my changes here:

https://github.com/litex-hub/litex-boards/compare/master...jiegec:litex-boards:vcu128?expand=1

https://github.com/enjoy-digital/litex/compare/master...jiegec:litex:vcu128?expand=1

https://github.com/enjoy-digital/litedram/compare/master...jiegec:litedram:vcu128?expand=1

enjoy-digital, 2023-04-24:
Awesome work @jiegec! Thanks for sharing the issues/process, this has been very useful to also understand your changes to the codebase. I also read your [blog post ](https://jia-je.translate.goog/hardware/2023/04/20/dram-topology-training/?_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=fr&_x_tr_pto=wapp) to understand your changes (If you don't mind, I added a link to it in listed it to https://github.com/enjoy-digital/litex/wiki/Tutorials-Resources#useful-resources).

The VCU128 and Clamshell LiteDRAM/LiteX support seems fine and we'll be able to merge them. Before doing so, I just want to think a bit about the Clamshell support to LiteDRAM and see how we could eventually reduce changes in the codebase. I'll get back to you in a few days.

enjoy-digital, 2023-05-25:
The different PRs are now merged, thanks a lot!

closed 2023-05-25 by enjoy-digital