litex-hub/litex-boards #522: Nitefury board PCIe clock setting

myftptoyman, 2023-08-14:
I setup one Nitefury board and running [firesim](https://github.com/firesim/firesim) and connect the virtual bridge by PCIe is pass.
But when I setup the [litex-boards](https://github.com/litex-hub/litex-boards) by 
python3 -m litex_boards.targets.sqrl_acorn  --with-pcie --uart-name=crossover --build --load --driver
It can't list on lspci.

I also check the PCIE by [NiteFury-and-LiteFury](https://github.com/RHSResearchLLC/NiteFury-and-LiteFury).
The first time compile by vVvado is also did not work. 
I reading the [Sample Project](https://github.com/RHSResearchLLC/NiteFury-and-LiteFury/blob/master/Sample-Projects/Project-0/FPGA/readme.md) the LOC constraint will fix by Xilinx generated XDC file. So follow the guide put the generated XDC file as NORMAL processing order. Finally it is working.

So I modify the litex generated XDC file to the same as NiteFury
**sqrl_acorn.xdc**

```
set_property LOC GTPE2_CHANNEL_X0Y7 [get_cells {pcie_s7/inst/inst/Top_xdma_0_0_pcie2_to_pcie3_wrapper_i/pcie2_ip_i/inst/inst/gt_top_i/pipe_wrapper_i/pipe_lane[3].gt_wrapper_i/gtp_channel.gtpe2_channel_i}]

set_property LOC GTPE2_CHANNEL_X0Y6 [get_cells {pcie_s7/inst/inst/Top_xdma_0_0_pcie2_to_pcie3_wrapper_i/pcie2_ip_i/inst/inst/gt_top_i/pipe_wrapper_i/pipe_lane[0].gt_wrapper_i/gtp_channel.gtpe2_channel_i}]

set_property LOC GTPE2_CHANNEL_X0Y5 [get_cells {pcie_s7/inst/inst/Top_xdma_0_0_pcie2_to_pcie3_wrapper_i/pcie2_ip_i/inst/inst/gt_top_i/pipe_wrapper_i/pipe_lane[2].gt_wrapper_i/gtp_channel.gtpe2_channel_i}]

set_property LOC GTPE2_CHANNEL_X0Y4 [get_cells {pcie_s7/inst/inst/Top_xdma_0_0_pcie2_to_pcie3_wrapper_i/pcie2_ip_i/inst/inst/gt_top_i/pipe_wrapper_i/pipe_lane[1].gt_wrapper_i/gtp_channel.gtpe2_channel_i}]
```

**sqrl_acorn.tcl**

```
create_ip -vendor xilinx.com -name pcie_7x -module_name pcie_s7
set obj [get_ips pcie_s7]
set_property -dict [list \
CONFIG.Bar0_Scale {Megabytes} \
CONFIG.Bar0_Size {1} \
CONFIG.Buf_Opt_BMA {True} \
CONFIG.Component_Name {pcie} \
CONFIG.Device_ID {7024} \
CONFIG.Interface_Width {128_bit} \
CONFIG.Link_Speed {5.0_GT/s} \
CONFIG.Max_Payload_Size {512_bytes} \
CONFIG.Maximum_Link_Width {X4} \
CONFIG.PCIe_Blk_Locn {X0Y0} \
CONFIG.Ref_Clk_Freq {100_MHz} \
CONFIG.Trans_Buf_Pipeline {None} \
CONFIG.Trgt_Link_Speed {4'h2} \
CONFIG.User_Clk_Freq {125} \
CONFIG.Legacy_Interrupt {None} \
CONFIG.IntX_Generation {False} \
CONFIG.MSI_64b {False} \
CONFIG.Multiple_Message_Capable {1_vector} \
] $obj
synth_ip $obj

set_property PROCESSING_ORDER NORMAL [get_files pcie_s7-PCIE_X0Y0.xdc]
```

But it is don't work on litex-board

I double check the NiteFury-and-LiteFury's PCIE clock structure.
I found out the the MMCM's reference clock is wrong on litex-board

**sqrl_acorn.v**
assign s7pciephy_clkin = s7pciephy_pipe_txoutclk_bufg;

**NiteFury-and-LiteFury**
assign s7pciephy_clkin = s7pciephy_pcie_refclk;

But it also don't work.
I check the link_status by crossover uart and bridge.
and found the MMCM clock is never locked.
and found out the PCIe's pcie_clkreq_l is not connected on sqrl_acorn.

So Add pcie_acorn one port pcie_clkreq_l and connect 1'b0 to always request a clock on the PCIE bus to train the MMCM to lock the clock phase.

**sqrl_acorn.v**
    output wire          pcie_clkreq_l
    assign pcie_clkreq_l = 1'b0;

**sqrl_acorn.tcl**
set_property PACKAGE_PIN G1 [get_ports {pcie_clkreq_l}]
set_property IOSTANDARD LVCMOS33 [get_ports {pcie_clkreq_l}]

Finally, it is work.

 ❯ ./litepcie_util dma_test
[> DMA loopback test:
---------------------
DMA_SPEED(Gbps)	TX_BUFFERS	RX_BUFFERS	DIFF	ERRORS
         10.29	     31553	     31425	 128	     0
         10.30	     63137	     62977	 160	     0
         10.29	     94689	     94561	 128	     0
         10.29	    126241	    126113	 128	     0
         10.30	    157825	    157697	 128	     0
         10.29	    189377	    189249	 128	     0
         10.30	    220961	    220801	 160	     0
         10.29	    252513	    252385	 128	     0
         10.31	    284129	    283969	 160	     0
         10.30	    315713	    315585	 128	     0
DMA_SPEED(Gbps)	TX_BUFFERS	RX_BUFFERS	DIFF	ERRORS
         10.30	    347297	    347169	 128	     0
         10.29	    378849	    378721	 128	     0
         10.30	    410433	    410305	 128	     0
         10.29	    441985	    441857	 128	     0
         10.31	    473601	    473441	 160	     0
         10.29	    505153	    505025	 128	     0
         10.31	    536769	    536609	 160	     0
         10.30	    568353	    568225	 128	     0
         10.30	    599937	    599809	 128	     0
         10.30	    631521	    631393	 128	     0

My environment:
CPU: i5-12500
Ubutun 22.04
ASROCK DeskMini B660
64G DDR4
1T Samsung 980 pro
Nitefury install on first M.2 (PCIE 5.0 port)

The problem maybe is my board's PCIE timing is very wrong, the correct constraint and clock feedback are very critical.

Hope the experiment can help used the same board as Nitefury.

enjoy-digital, 2023-09-06:
Thanks @myftptoyman, this indeed seems specific to your machine, but I'll make sure to integrate your change in the repo.

enjoy-digital, 2023-09-27:
@myftptoyman: Your fix has been integrated with https://github.com/litex-hub/litex-boards/commit/fd6aee0250898a77f02f28e5d5afd362a70e1b06, thanks! For the PCIe lane mapping, this should be fine since validated on other machines.

closed 2023-09-27 by enjoy-digital