litex-hub/litex-boards #530: Bitstream loading on ECPIX-5 board needs --cable ft4232

suarezvictor, 2023-09-20:
openFPGALoader needs that setting, if not the cable defaults to ft2232 and the onboard FTDI device is not found

trabucayre, 2023-09-21:
ecpix5 has now two variants:
- new (*R03*) based on **ft4232**
- previous based on **ft2232**

I have pushed a new board definition for *R03* but it make sense (at least in a first time / until a new openFPGALoader' s release) to add a `--revision` to override the cable explicitly.

suarezvictor, 2023-09-22:
Good, I didn't know there were two revisions, many thanks for sharing the info

enjoy-digital, 2026-10-21:
This is now handled with `--version r03` on the `lambdaconcept_ecpix5` target, which selects the `ecpix5_r03` openFPGALoader board (FT4232). Thanks!

closed 2026-10-21 by enjoy-digital