litex-hub/litex-boards #534: Unable to replicate Analogue Pocket

agg23, 2023-09-29:
Rudimentary support for Analogue Pocket was added in 3df677cfeb21f9a07aca425abe74db71d502aa56, and I wanted to see if I could start experimenting with LiteX in this context. After a period of discovering that most of the LiteX tools (at least for Intel) don't run on Windows, I was able to fully build and program the Cyclone V using the USB Blaster (in Linux).

However, opening a console (through `python3 litex_term.py jtag --jtag-config openocd_usb_blaster.cfg`) results in only the characters `sL5DdSMmkekro` being printed. That seems to be a common string in LiteX serial output, but I was unable to determine what it means. Some poking around confirmed that it was coming from the device (and assumedly the program), so something is running and JTAG is functioning in some manner.

Passing `--kernel bios.bin` to `litex_term` claims to receive a request from the device, but it fails shortly after:
```
[LITEX-TERM] Received firmware download request from the device.
[LITEX-TERM] Uploading build/analog_pocket/software/bios/bios.bin to 0x40000000 (24872 bytes)...
[LITEX-TERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LITEX-TERM] Got unexpected response from device 'b'E''
```

Following https://github.com/enjoy-digital/litex/wiki/Use-Host-Bridge-to-control-debug-a-SoC, I started `litex_server --jtag --jtag-config openocd_usb_blaster.cfg` and attempted to connect to it directly in Python using `RemoteClient`. Attempting to perform any reads resulted in a timeout (writes didn't, but I just assume there's no verification of a successful write).

----

All of this makes me suspect that I don't actually have a working JTAG connection, but I have no idea how to validate that. The bitstream flash is significantly slower than Quartus, but that might simply be a difference in configuration.

I apologize if I've missed something obvious, but at this point I've spent a long time debugging this issue and I figure it's probably time to create an issue. If nothing else, someone can learn from my misfortune.

AndrewD, 2023-09-30:
This string `sL5DdSMmkekro` is part of the normal BIOS boot, you should have seen the last bit of the boot messages if you waited a few more seconds.
I also replied to your message on discord.

enjoy-digital, 2023-10-02:
Hi @agg23,

if you have the developer cart bridge, as discussed on Discord, I would recommend using it as UART to start with:
- Make sure that the serial pins defined here: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/analog_pocket.py#L18-L23 match the ones of the dev cart bridge and modify if not.
- Build and load with design with `./analog_pocket.py --build --load`.
- Open LiteX Term: `litex_term /dev/ttyUSBX`

enjoy-digital, 2023-10-02:
@agg23: OK, that's strange. Would you mind testing this bitstream (with JTAG-UART) to see if it behaves similarly? (Just want to rule out an RISC-V GCC eventual issue):  [analog_pocket_2023_10_02.zip](https://github.com/litex-hub/litex-boards/files/12783460/analog_pocket_2023_10_02.zip)

Also just for info, I was testing this with the Pong FPGA core previously loaded/running. Not sure it should impact behavior, but could still be worth testing if still not working.

agg23, 2023-10-02:
> @agg23: OK, that's strange. Would you mind testing this bitstream (with JTAG-UART) to see if it behaves similarly? (Just want to rule out an RISC-V GCC eventual issue): [analog_pocket_2023_10_02.zip](https://github.com/litex-hub/litex-boards/files/12783460/analog_pocket_2023_10_02.zip)

That is bizarre. Your build works correctly. It prints the BIOS startup message, and responds to my commands.

Installed GCC RISC-V is:
```
binutils-riscv64-unknown-elf/jammy,now 2.35.1-0ubuntu1 amd64 [installed,automatic]
gcc-riscv64-unknown-elf/jammy,now 10.2.0-0ubuntu1 amd64 [installed]
```

agg23, 2023-10-03:
I don't know if this was my fault or not, but it seems like that LiteX build environment was just messed up somehow. I literally had just created it, as it was created for the purpose of testing this target, and I don't think I did anything unusual, just install via:
```bash
litex_setup.py --init --install --config=standard
```

In any case, my second environment, which contained a scoped set of the LiteX packages as Git submodules, seems to be producing working builds.

Since this is highly likely not a symptom of the target, I will close this issue, though there may be an issue hiding here with LiteX installs.

closed 2023-10-03 by agg23