litex-hub/litex-boards #549: Sipeed Tang Mega 138k : SDRAM test failed.

agv100, 2023-11-22:
Hello,
I'm trying to build litex bytestream with DDR3 support.
Previously, on-board DDR3 wa checked with GAO/simple example in sipeed github examples, so, should be operational.

But, when I load litex bitstream, build with --with-ddr3 support, I see, initially ddr3 could be seen:
```       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Nov 21 2023 15:36:50
 BIOS CRC passed (dabbc66a)

 LiteX git sha1: 02277680

--=============== SoC ==================--
CPU:            VexRiscv @ 50MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128.0KiB
SRAM:           8.0KiB
SDRAM:          1.0GiB 32-bit @ 200MT/s (CL-6 CWL-5)
MAIN-RAM:       1.0GiB

--========== Initialization ============--
```

But it's tests are failed:
```
litex> sdram_init
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |00000000| delays: -
  m0, b01: |00000000| delays: -
  m0, b02: |00000000| delays: -
  m0, b03: |00000000| delays: -
  best: m0, b00 delays: -
  m1, b00: |00000000| delays: -
  m1, b01: |00000000| delays: -
  m1, b02: |00000000| delays: -
  m1, b03: |00000000| delays: -
  best: m1, b00 delays: -
  m2, b00: |00000000| delays: -
  m2, b01: |00000000| delays: -
  m2, b02: |00000000| delays: -
  m2, b03: |00000000| delays: -
  best: m2, b00 delays: -
  m3, b00: |00000000| delays: -
  m3, b01: |00000000| delays: -
  m3, b02: |00000000| delays: -
  m3, b03: |00000000| delays: -
  best: m3, b00 delays: -
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
  bus errors:  256/256
  addr errors: 0/8192
  data errors: 524288/524288
Memtest KO
```

And, when I try to access memory manually, it looks not working:
```
litex> mem_write 0x40000010 0xff 1

litex> mem_read 0x40000010 1
Memory dump:
0x40000010  00   
```

What could be a problem?

trabucayre, 2023-11-23:
Hello,
Yes DDR3 is not working and is a WIP.
Sipeed example uses Gowin's DDR3 encrypted IP. But since Gowin doesn't provides real documentation about primitives required to add this support it's more difficult/long to have a working support.
If you need SDRAM it's possible to plug a Mister SDRAM extension into SDRAM dock's connector.

enjoy-digital, 2024-01-07:
@agv100: We've planned some to work on this in the following weeks/months. This will probably happen soon but we are working on others features in priority on other features for now.

enjoy-digital, 2024-01-14:
While continuing this work, it could be worth checking if a specific LOC constraint is required to get the DDR3 working, as it is for the AE350 SOC:
[image]

briansune, 2024-01-29:
@enjoy-digital 
@trabucayre 

I would like to give it a try on the modified board.
The background is because before 19.9 is released the DDR3 IP cannot run AE350 with non-Micron DDR.
You can read more on here about what we had done:
https://github.com/briansune/Tang-Mega-138K-AE350-Demo

If such design is based on the default official board hardware then my current hardware is 100% aligned.

agv100, 2024-08-22:
Hello,
Just found new commit, adding both sipeed and mister SDRAMs. 
So, if I build (without today commit on litedram/litedram/frontend/wishbone.py as it fails on assertion    assert wishbone.addressing == "byte" ), with options:
```./litex-boards/litex_boards/targets/sipeed_tang_mega_138k_pro.py  --with-sdram  --sdram-model sipeed --cpu-type=gowin_ae350 --build
```

I still can see sdram init failed, but, when I manually read-write upto 4-byte words, it works well:

```litex> sdram_init
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
  bus errors:  128/256
  addr errors: 0/8192
  data errors: 524288/524288
Memtest KO

litex> mem_write 0x40100000 0xabcd

litex> mem_read 0x40100000
Memory dump:
0x40100000  cd ab 00 00                                      ....

litex>
```

Anything I have to do to get it fully working?

briansune, 2024-08-22:
@agv100 

I think you had an misunderstand on what the last commission.

Read and think is what you need. The commit is for external SDRAM PMOD.
Not the default DDR3 memory.

End of story.

Bests,
Brian

TONYHEGUAGUA, 2024-10-08:
> Hello, I'm trying to build litex bytestream with DDR3 support. Previously, on-board DDR3 wa checked with GAO/simple example in sipeed github examples, so, should be operational.
>
> But, when I load [...]

hello ,i recently try to run litex on sipeed tang mega 138k pro ,  i'm now facing the exact same problem as you do . Have you figured out how to solve this problem yet ? 

[image]
[image]

quanweizhang, 2024-10-24:
您好，请问你们怎么解决这个问题的呢？

2jack5657, 2024-10-30:
> Hello, 你好 Yes DDR3 is not working and is a WIP.是的，DDR3 不工作，是一个 WIP。 Sipeed example uses Gowin's DDR3 encrypted IP. But since Gowin doesn't provides real documentation about primitives required to add this support it's more difficult/long to have a working support.Sipeed 示例使用 Gowin 的 DDR3 加密 IP。但是由于 Gowin 没有提供有关添加此支持所需的原始变量的真实文档，因此拥有工作支持更加困难/漫长。 If you need SDRAM it's possible to plug a Mister SDRAM extension into SDRAM dock's connector.如果您需要 SDRAM，可以将 Mister SDRAM 扩展插入 SDRAM 扩展坞的连接器。

I recently bought sipeed SDRAM Plugin,but when i plug in ,my Soc seems not to reconize the RAM storage correctly. The mod is 32MiB*2 ,but it only can be recognized 32MiB,and sdram_init is also fail. I use Tang mega 138k, the platform and target profile are changed from Tang mega 138k pro.
[image]
[image]

fjullien, 2026-08-29:
Hi @enjoy-digital, did you make any progress on that?

I have a variant of that configuration. My module is a 60K. This is what I get (PR #759 included):
```

--=========== Initialization ===========--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |00000000| delays: -
  m0, b01: |00000000| delays: -
  m0, b02: |00000000| delays: -
  m0, b03: |00000000| delays: -
  best: m0, b02 delays: -
  m1, b00: |00000000| delays: -
  m1, b01: |00000000| delays: -
  m1, b02: |00000000| delays: -
  m1, b03: |00000000| delays: -
  best: m1, b00 delays: -
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
  bus errors:  2/256
  addr errors: 6144/8192
  data errors: 470464/524288
Memtest KO
Memory initialization failed
```

enjoy-digital, 2026-09-09:
Hi @fjullien,

it should now be good with a LiteDRAM, LiteX and LiteX-Boards update :)

fjullien, 2026-09-09:
Thank you for working on this!

At 50MHz, 1:2 rate, it's working. Read leveling numbers are weird.

```
--================ SoC =================--
CPU:            VexRiscv @ 50MHz
BUS:            wishbone 32-bit data/32-bit addr
CSR:            32-bit data big ordering
ROM:            128.0KiB
SRAM:           8.0KiB
SDRAM:          512.0MiB 16-bit @ 200MT/s (CL-6 CWL-6)
MAIN RAM:       512.0MiB

--=========== Initialization ===========--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |11111111111111111111111111111111| delays: 63+-63
  m0, b03: |00000000000000000000000000000000| delays: -
  best: m0, b02 delays: 63+-63
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |11111111111111111111111111111111| delays: 63+-63
  m1, b03: |00000000000000000000000000000000| delays: -
  best: m1, b02 delays: 63+-63
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 21.4MiB/s
   Read speed: 7.8MiB/s
--================ Boot ================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
No boot medium found

--============== Console ===============--

litex> mem_test 0x40000000 0x01000000
Memtest at 0x40000000 (16.0MiB)...
  Write: 0x40000000-0x41000000 16.0MiB
   Read: 0x40000000-0x41000000 16.0MiB
Memtest OK
litex> mem_test 0x50000000 0x01000000
Memtest at 0x50000000 (16.0MiB)...
  Write: 0x50000000-0x51000000 16.0MiB
   Read: 0x50000000-0x51000000 16.0MiB
Memtest OK
litex> mem_test 0x5F000000 0x01000000
Memtest at 0x5f000000 (16.0MiB)...
  Write: 0x5f000000-0x60000000 16.0MiB
   Read: 0x5f000000-0x60000000 16.0MiB
Memtest OK
litex> mem_test 0x40000000 0x20000000
Memtest at 0x40000000 (512.0MiB)...
  Write: 0x40000000-0x60000000 512.0MiB
   Read: 0x40000000-0x60000000 512.0MiB
Memtest OK
litex>
```
At 100MHz, 1:4, I get some errors:

```
--================ SoC =================--
CPU:            VexRiscv @ 100MHz
BUS:            wishbone 32-bit data/32-bit addr
CSR:            32-bit data big ordering
ROM:            128.0KiB
SRAM:           8.0KiB
SDRAM:          512.0MiB 16-bit @ 800MT/s (CL-6 CWL-5)
MAIN RAM:       512.0MiB

--=========== Initialization ===========--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b02 delays: -
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b02 delays: -
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
  bus errors:  0/256
  addr errors: 340/8192
  data errors: 25244/524288
Memtest KO
Memory initialization failed
--============== Console ===============--

litex>
```
It also doesn't work at 25MHz 1:4.
I'll check my target file more carefully.

enjoy-digital, 2026-09-10:
@fjullien: Thanks for the feedback, I'll try to improve the leveling results yes. I'm testing with the Tang Mega 138K and will see if I also have the Console 60k to test it.

enjoy-digital, 2026-09-10:
@fjullien: I have the console 60k, so will also test with it and try to understand/fix.

trabucayre, 2026-09-10:
Hi all
In my side both 50MHz/1:2 and 100MHz/1:4 works on sipeed tang mega 138k with the same read-leveling.

For Sipeed tang console I have:

For 50MHz/1:2:

```bash
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2026 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Sep 10 2026 09:25:18
 BIOS CRC passed (f8df0b8a)

 LiteX git sha1: 9adcf76be

--================ SoC =================--
CPU:            VexRiscv @ 50MHz
BUS:            wishbone 32-bit data/32-bit addr
CSR:            32-bit data big ordering
ROM:            128.0KiB
SRAM:           8.0KiB
SDRAM:          512.0MiB 16-bit @ 200MT/s (CL-6 CWL-6)
MAIN RAM:       512.0MiB

--=========== Initialization ===========--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |11111111111111111111111111111111| delays: 63+-63
  m0, b03: |00000000000000000000000000000000| delays: -
  best: m0, b02 delays: 63+-63
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |11111111111111111111111111111111| delays: 63+-63
  m1, b03: |00000000000000000000000000000000| delays: -
  best: m1, b02 delays: 63+-63
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 21.4MiB/s
   Read speed: 7.8MiB/s
--================ Boot ================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
No boot medium found

--============== Console ===============--
```

and for 100MHz/1:4:

```bash
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2026 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Sep 10 2026 09:30:07
 BIOS CRC passed (9e47a848)

 LiteX git sha1: 9adcf76be

--================ SoC =================--
CPU:            VexRiscv @ 100MHz
BUS:            wishbone 32-bit data/32-bit addr
CSR:            32-bit data big ordering
ROM:            128.0KiB
SRAM:           8.0KiB
SDRAM:          512.0MiB 16-bit @ 800MT/s (CL-6 CWL-5)
MAIN RAM:       512.0MiB

--=========== Initialization ===========--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |11111111111111111111111111111111| delays: 63+-63
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b02 delays: 63+-63
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |11111111111111111111111111111111| delays: 63+-63
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b02 delays: 63+-63
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 68.6MiB/s
   Read speed: 23.4MiB/s
--================ Boot ================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
No boot medium found

--============== Console ===============--
```

trabucayre, 2026-09-10:
@fjullien you uses the `sipeed_tang_console.py` or code adapted from https://github.com/trabucayre/litex-boards/tree/merge_60k_in_mega138k ?

enjoy-digital, 2026-09-10:
Thanks for the test/feedback @trabucayre!

enjoy-digital, 2026-09-10:
@fjullien: I reproduced and did other improvements on top. Could you do another test and provide feedback? Thanks.

fjullien, 2026-09-11:
@trabucayre I use the code adapted to my Tang Mega 60K.
@enjoy-digital now it works:

```
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2026 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Sep 11 2026 07:52:34
 BIOS CRC passed (03ff31b9)

 LiteX git sha1: 743825d3f

--================ SoC =================--
CPU:            VexRiscv @ 100MHz
BUS:            wishbone 32-bit data/32-bit addr
CSR:            32-bit data big ordering
ROM:            128.0KiB
SRAM:           8.0KiB
SDRAM:          512.0MiB 16-bit @ 800MT/s (CL-6 CWL-5)
MAIN RAM:       512.0MiB

--=========== Initialization ===========--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write DQ-DQS training:
m0: |11111111000000000000000000000000| delays: 29+-29
m1: |11111111000000000000000000000000| delays: 30+-30
Read leveling:
  m0, b02: |11111111000000000000000000000000|
  best: m0, b02 delays: 28+-28
  m1, b02: |11111110000000000000000000000000|
  best: m1, b02 delays: 26+-26
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 68.6MiB/s
   Read speed: 23.4MiB/s
--================ Boot ================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
No boot medium found

--============== Console ===============--

litex>
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2026 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Sep 11 2026 07:59:08
 BIOS CRC passed (2da124d5)

 LiteX git sha1: 743825d3f

--================ SoC =================--
CPU:            VexRiscv @ 50MHz
BUS:            wishbone 32-bit data/32-bit addr
CSR:            32-bit data big ordering
ROM:            128.0KiB
SRAM:           8.0KiB
SDRAM:          512.0MiB 16-bit @ 200MT/s (CL-6 CWL-6)
MAIN RAM:       512.0MiB

--=========== Initialization ===========--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b02: |11111111111111111111111111111111|
  best: m0, b02 delays: 127+-127
  m1, b02: |11111111111111111111111111111111|
  best: m1, b02 delays: 127+-127
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 21.4MiB/s
   Read speed: 7.8MiB/s
--================ Boot ================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
No boot medium found

--============== Console ===============--

litex>
```
Thanks!

enjoy-digital, 2026-10-21:
Closing since SDRAM is now reported working, thanks! Feel free to reopen if issues remain on the Tang Mega 138K.

closed 2026-10-21 by enjoy-digital