litex-hub/litex-boards #554: limesdr_mini_v2 loading firmware: [LITEX-TERM] Got unexpected response from device 'b'\n''

vbuitvydas, 2023-12-22:
Hi,

I am trying to load firmware to LimeSDR-Mini v2 board. I have generated and loaded gw for limesdr_mini_v2 target. Also I have taken firmware from [lab004](https://github.com/litex-hub/fpga_101/tree/master/lab004/firmware), removed led_test and display_test and compiled firmware, but I cannot load it. Can anyone help me to understand what is wrong here?

**Issue:**
I am getting `[LITEX-TERM] Got unexpected response from device 'b'\n''` while loading firmware. 

[image]

**Steps to reproduce issue:**
./limesdr_mini_v2.py --csr-csv=csr.csv --build --load

litex_server --jtag --jtag-config=openocd_limesdr_mini_v2.cfg
litex_term crossover --kernel ./build/limesdr_mini_v2/firmware/firmware.bin

enjoy-digital, 2024-01-07:
@vbuitvydas: Uploading firmware over JTAG was not well tested/reliable in the past and we made some work on this recently. If you are not using upstream version of LiteX, it could be worth testing with it.  If still not yet working, I'll have a look.

vbuitvydas, 2024-01-08:
thanks @enjoy-digital for your reply, 

I was using Litex [7062e3379fdd80ac0523390ca46c87906b8c68ff](https://github.com/enjoy-digital/litex/tree/7062e3379fdd80ac0523390ca46c87906b8c68ff), and updated it to the latest [91fbc79ac2e0af4e9edf3a3fe423f98fb7bdeffb](https://github.com/enjoy-digital/litex/tree/91fbc79ac2e0af4e9edf3a3fe423f98fb7bdeffb). 

I am getting similar results:
[image]

trabucayre, 2024-04-11:
Could you retry with:
```
./limesdr_mini_v2.py --integrated-main-ram-size 0x8000 --load --build --uart-name=jtag_uart
litex_bare_metal_demo --build-path build/limesdr_mini_v2
litex_term jtag --jtag-config=openocd_limesdr_mini_v2.cfg --kernel demo.bin
```

Seems to have a problem to load a firmware with `jtagbone` + `crossover`, but it's working when using `jtag_uart`

vbuitvydas, 2024-04-12:
> Could you retry with:
>
> ```
> ./limesdr_mini_v2.py --integrated-main-ram-size 0x8000 --load --build --uart-name=jtag_uart
> litex_bare_metal_demo --build-path build/limesdr_mini_v2
> litex_term jtag [...]

Tried with following commands and it is working now. Thanks!

trabucayre, 2024-04-12:
Thanks for your feedback!

vbuitvydas, 2024-04-16:
Hi @trabucayre,  

with latest litex updates firmware loading  using jtag_uart is no longer working. I followed same steps as mentioned above litex_term gets stuck:
[image]

trabucayre, 2024-04-16:
Indeed.
It's a regression with commit 6885770e4736b234e65f57840ff8f74a1165a7e0.
After reverting this commit jtag_uart is working.