litex-hub/litex-boards #558: Sipeed Tang primer 20k : usb host test failed.

hai4john, 2024-01-10:
I had Sipeed Tang primer 20k with dock, USB host on hardware, but test filed.
I saw the source code at targets/sipeed_tang_primer_20k.py，found no usb host support。

hai4john, 2024-01-11:
After looking at the code for the USB part in the liteX core, I found that only the 'Differential Pair mode' USB OHCI was implemented, and the '8-bit parallel mode' was not implemented.

enjoy-digital, 2024-01-14:
Hello @hai4john,

for now we indeed only support USB OHCI, it would be possible to support ULPI but support hasn't been done yet.

hai4john, 2024-01-15:
You may not clearly distinguish between USB controller and PHY (physical layer interface). The current code seems to implement controller + phy, not a separate controller.

enjoy-digital, 2024-01-15:
The current supported solution is indeed full integrated and only USB 1.1 compliant . I just meant that that Tang primer 20k dock has a ULPI chip and we don't currently have built-in solution for it.

hai4john, 2024-01-15:
@enjoy-digital, I understand what you mean. Thank you very much for your reply. I look forward to the implementation of USB host on Tang primer 20k dock.

> The current supported solution is indeed full integrated and only USB 1.1 compliant . I just meant that that Tang primer 20k dock has a ULPI chip and we don't currently have built-in solution for it.