litex-hub/litex-boards #565: Sipeed Tang Primer 25k no serial out after successful flash

blark, 2024-02-15:
Was hoping someone could help me understand why after a successful build, and flash I am not getting anything from the UART of my 25k.

I had a similar problem with my 9k, and did some digging and found recommendations in issues to comment out a few lines--which got it working. However, I don't see similar lines in the 25k board files, I did play around a bit but am a n00b with LiteX so I'm a bit lost.

I know the flash worked fine as the LED chaser works, which is the same thing that happened on the 9k. I'm just once again not getting anything when I connect at 115200 baud. Thanks in advance for your assistance.

trabucayre, 2024-02-19:
Yes. This issue is unrelated to the 9k.
Here the firmware executed by the riscv is embedded into the bitstream so no need to write the firmware into the flash.

First of all: could you copy/paste dmesg displayed when you plug the board. I have seen some issues with recent usb interface firmware.

blark, 2024-02-22:
Thanks for the reply. I'm currently travelling but when I get back home next week I'll send the details. 

The 9k serial terminal worked fine (after I commented those lines), does the 25k use a different driver?

trabucayre, 2024-02-23:
There is two unrelated issue here:
- 9k needs to have to a riscv firmware written into external flash and the CPU execute code directly from this device
- 25k has a more classic approach (riscv firmware is stored into BRAM.

9k and 25k have different usb<->serial converters. These chip are microcontroller with a firmware (not the one built by LiteX) to mimic FT2232 chip. I suspect some regressions with current firmware shipped with 25k (mine board has an older firmware without this type of issue). This why I need to confirm/infirm my assumption with your 25k interface version.