litex-hub/litex-boards #574: 5a-75b revision 8.2

haubke, 2024-03-24:
hello,
i have tried to build multiple projects using liteeth on a 5a-75b 8.2 board.(liteeth, led cube from lucy and others.  unfortunatly i do not own a scope. and i have trouble getting the ethernet to work, i have found some solutions/problems and i have tried all of them. 
even tried to make one with an risc v core(litex).

has anybody got an working example with ethernet for the 5a-75b V8.2? or even just the tx/rx delay timings? i have been busy for over 4 weeks now, with no succes. is it even possible?  if i only have 100megs it will be sufficient for my usecase. 

i want to build a led screen that is capable of displaying DDP data directly from xlights,  any suggestions?

haubke, 2024-04-05:
Update: i was able to ping the board.  Only by using  litecolor.  So it is working. However when i try to use liteeth. Pinging the board is not working.  I'm trying to get a working liteeth core. I Will keep this page updated when i find something.

enjoy-digital, 2024-06-19:
Hi @haubke,

can you share a link  to the design that worked on your hardware? and link or command to the one that does not work? I'll try to understand the differences.

kscz, 2025-01-04:
Is there anything I could do to help make this a reality? I unfortunately bought v8.2 boards without realizing they were not yet supported, so I'm now motivated to help fix the litex repos to support what I have!

I see it looks like the pinouts have largely been documented - https://github.com/q3k/chubby75/blob/master/5a-75b/hardware_V8.2.md - should I just try and walk through the tables and generate matching values in `platforms/colorlight_5a_75b.py` ?

enjoy-digital, 2025-01-04:
@kscz: If the boards has been documentated, you can indeed try to update the platform and test. If issues, you can ask here, we'll try to help.

kscz, 2025-01-04:
I am attempting to use https://github.com/enjoy-digital/colorlite to test out my changes, but I'm getting an error: `Cannot extract CSR name from code, need to specify`

I tried getting rid of all my changes and creating a clean environment, and I'm getting errors just trying to make `./colorlite.py --build` work at all

Is there something I need to change to get your "colorlite" repository working?

kscz, 2025-01-12:
Anything I can do to help move along the PR?

closed 2025-01-14 by enjoy-digital

haubke, 2025-01-20:
srr for the late reply, dont recall why i could not get it to work . as of now i'm using lucyrausch liteeth core(from colorlite led cube) on a version 8.2 board.  (if oyu plan on using this, there is a 32bit bus for udp_source_data, in reality its only 7 bits wide.). 

i also could not get migen to install.  so i did not build the latest core.

PR #635 (2025-01-14): Add in colorlight v8.2 board support
Based on https://github.com/q3k/chubby75/blob/master/5a-75b/hardware_V8.2.md it appears that the pinout v8.0 is the same as the v8.2, the only difference is the FPGA - a Lattice ECP5 LFE5U-25F-7BG256I

The SDRAM for the 8.0 and 8.2 is the same, but the CLS was not being correctly set, this commit also corrects this.

Tested on a v8.2 board using https://github.com/enjoy-digital/colorlite

Fixes #574