litex-hub/litex-boards #594: zedboard can't boot up correctly

phillippZZ, 2024-06-23:
Hi, everyone.

I built a naxriscv bitstream and loaded it to zedboard. When I used the default linux image and other configurations file in a sdcard, I observed something on my monitor and the board had a few indicators powered on. 
[image]
But when I put into the sdcard with files downloaded from "https://github.com/litex-hub/linux-on-litex-vexriscv/files/8331338/linux_2022_03_23.zip", the board only has power indicator lighting. I can't see anything on monitor or data transfered via uart on PuTTY

[image]

Is it because the image files don't use uart or I did something wrong?

trabucayre, 2024-06-23:
I suppose the first image used was a stock firmaware for digilent? If true: it uses the PS (processor) and the PL(FPGA) is dedicated to extends features (VGA/HMDI).
With the second image from linux-on-vexriscv, PS part isn't used but SDCARD, UART, SDRAM and ethernet aren't available from PL. There is PR to access PS resources from PL but we have to review/merge it before.
So If you have to add PMOD extensions for SDCARD/UART/... and to modify the target accordingly.

phillippZZ, 2024-06-24:
Thank you very much.
Some of the concepts are new to me, I'll have some research and test for now.

phillippZZ, 2024-06-24:
hi, @trabucayre

I added the following code to _io in "litex_boards/platforms/digilent_zedboard.py":
>("serial", 0,
    Subsignal("tx", Pins("D11")),  # MIO[48] for TX
    Subsignal("rx", Pins("C14")),  # MIO[49] for RX
    IOStandard("LVCMOS33")         # Assuming 3.3V logic levels
),

The ID of the pins come from "https://digilent.com/reference/_media/zedboard:zedboard_ug.pdf" as shown in the attached image.
[image]

Then I got the following error from compiling:
>ERROR: [DRC UCIO-1] Unconstrained Logical Port: 2 out of 11 logical ports have no user assigned specific location constraint (LOC). This may cause I/O contention or incompatibility with the board power or connectivity affecting performance, signal integrity or in extreme cases cause damage to the device or the components to which it is connected. To correct this violation, specify all pin locations. This design will fail to generate a bitstream unless all logical ports have a user specified site LOC constraint defined.  To allow bitstream creation with unspecified pin locations (not recommended), use this command: set_property SEVERITY {Warning} [get_drc_checks UCIO-1].  NOTE: When using the Vivado Runs infrastructure (e.g. launch_runs Tcl command), add this command to a .tcl file and add that file as a pre-hook for write_bitstream step for the implementation run.  Problem ports: serial_rx, and serial_tx.

trabucayre, 2024-06-25:
*MIO* pins are pins dedicated for PS, and not available from PL.
To have an UART you have to uses a PMOD extension (warning one of them is connected to the PS).
In fact, without the PR mentioned zedboard (with only the PL) has not enough BRAM to be able to run GNU/Linux OS.

phillippZZ, 2024-06-25:
May I ask is it possible to connect PL to the UART of PS via some type of PMOD extension? Does it mean I have to write some code to build communication between PL and PS?
Or is it better to connect PL to an external UART via PMOD extension?

trabucayre, 2024-06-25:
Yes, currently the best way is to have an external UART interface connected to a PMOD. Again there is work to allows PL to access to PS controller (and consequently to be able to use UART interface without external peripherals, but we have to review/test see how to have a easiest solution.

closed 2024-06-26 by phillippZZ