litex-hub/litex-boards #600: Fomu does not build without CPU

jamesdiacono, 2024-08-06:
On my machine, the command `./kosagi_fomu.py --cpu-type None --build` gives
```
INFO:iCE40PLL:Creating iCE40PLL, SB_PLL40_CORE primitive.
INFO:iCE40PLL:Registering Single Ended ClkIn of 48.00MHz.
INFO:iCE40PLL:Creating ClkOut0 usb_12 of 12.00MHz (+-10000.00ppm).
INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2024-08-06 12:00:37)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : ice40-up5k-uwg30.
INFO:SoC:System clock: 12.000MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Controller ctrl added.
INFO:SoC:CPU None added.
INFO:SoC:CPU None adding IO Region 0 at 0x00000000 (Size: 0x100000000).
INFO:SoCBusHandler:io0 Region added at Origin: 0x00000000, Size: 0x100000000, Mode:  RW, Cached: False, Linker: False.
INFO:SoCBusHandler:Allocating Cached Region of size 0x00020000...
INFO:SoCRegion:Region size rounded internally from 0xffffffff to 0x100000000.
INFO:SoCBusHandler:psram Region allocated at Origin: 0x00000000, Size: 0x00020000, Mode:  RW, Cached:  True, Linker: False.
INFO:SoCBusHandler:psram added as Bus Slave.
INFO:SoCBusHandler:sram Region added at Origin: 0x00000000, Size: 0x00010000, Mode:  RW, Cached:  True, Linker:  True.
INFO:SoCBusHandler:main_ram Region added at Origin: 0x00010000, Size: 0x00010000, Mode:  RW, Cached:  True, Linker:  True.
INFO:SoCBusHandler:spiflash Region added at Origin: 0x10000000, Size: 0x00200000, Mode:  RW, Cached:  True, Linker: False.
INFO:SoCBusHandler:spiflash added as Bus Slave.
INFO:SoCBusHandler:rom Region added at Origin: 0x10060000, Size: 0x00008000, Mode:  RW, Cached:  True, Linker:  True.
INFO:SoC:CSR Bridge csr added.
ERROR:SoCBusHandler:Region overlap between psram and csr:
ERROR:SoCBusHandler:Origin: 0x00000000, Size: 0x00020000, Mode:  RW, Cached:  True, Linker: False
ERROR:SoCBusHandler:Origin: 0x00000000, Size: 0x00010000, Mode:  RW, Cached: False, Linker: False
```
and fails with a non-zero exit code.

Possibly related to https://github.com/enjoy-digital/litex/issues/2014.

I am using macOS 14.5, Python 3.9.6, and litex-boards@81209b9 (the latest).

enjoy-digital, 2024-08-28:
Thanks @jamesdiacono for reporting, this is fixed with https://github.com/litex-hub/litex-boards/commit/c5d1a252c5a4767ce3e55f0ea954f429a745f1f9.

closed 2024-08-28 by enjoy-digital

tcal-x, 2024-08-28:
Oh!   Maybe the Fomu Workshop can bump its dependencies now: https://github.com/im-tomu/fomu-workshop/tree/master/litex/deps

I'm sure a bit of work would still be involved...