litex-hub/litex-boards #605: Titanium Ti375 C529 support

Dolu1990, 2024-08-26:
Hi,

This is just to notify that i'm working on adding support for this dev kit : https://www.efinixinc.com/products-devkits-titaniumti375c529.html

Let's me know if you have done any related work previously ^^

Dolu1990, 2024-08-27:
Got the uart / sdcard / usb host / debian to work.

enjoy-digital, 2024-08-28:
Nice, thanks for sharing @Dolu1990! Not sure anyone started adding LiteX-Boards support to it.

Dolu1990, 2024-08-28:
WIP : https://github.com/Dolu1990/litex-boards/tree/ti375_c529

Also got HDMI to work (https://www.digikey.fr/fr/products/detail/efinix-inc/EFX-HDMI/17084519)

enjoy-digital, 2024-08-28:
Great!

enjoy-digital, 2024-09-11:
Done with https://github.com/litex-hub/litex-boards/pull/610, thanks @Dolu1990!

enjoy-digital, 2024-09-11:
@Dolu1990: You different PR have been merged, would you mind doing a test with upstream repos? (I also ordered a Ti375 to also be able to do some tests but the board is not yet there).

Dolu1990, 2024-09-11:
@enjoy-digital Just tried with litex + litex-boards repository.
Seems all good, constraints look good and it does run debian at 200 Mhz ^^

Thanks !

enjoy-digital, 2024-09-11:
Great, thanks for the feedback! We can probably close this issue since completed.

closed 2024-09-11 by enjoy-digital

Dolu1990, 2024-09-11:
For reference, here is the set of parameters for the incoming orconf demo running debian : 

--cpu-type=vexiiriscv --cpu-variant=debian --with-jtag-tap --with-sdcard --with-coherent-dma --with-ohci --vexii-video "name=video" --vexii-args="--fetch-l1-hardware-prefetch=nl --fetch-l1-refill-count=2 --fetch-l1-mem-data-width-min=128 --lsu-l1-mem-data-width-min=128 --lsu-software-prefetch --lsu-hardware-prefetch rpt --performance-counters 9 --lsu-l1-store-buffer-ops=32 --lsu-l1-refill-count 4 --lsu-l1-writeback-count 4 --lsu-l1-store-buffer-slots=4  --relaxed-div" --l2-bytes=524288 --sys-clk-freq 100000000 --cpu-clk-freq 200000000 --with-cpu-clk --bus-standard axi-lite --cpu-count=4

enjoy-digital, 2024-09-11:
Thanks, indeed useful to save it somewhere :)