litex-hub/litex-boards #624: Option --toolchain=openxc7 ignored for Genesys2

Mecrisp, 2024-12-10:
I am trying to generate the LiteX example SoC for a Kintex-7 based Digilent Genesys2 board with a fresh LiteX install, using the openXC7 toolchain. But it seems as if the option --toolchain=openxc7 is ignored.

`~/FPGA/LiteX/litex-boards/litex_boards/targets/digilent_genesys2.py --toolchain=openxc7 --output-dir=experiment-nogates --build --no-compile-gateware`

Results in a build_digilent_genesys2.sh file to launch Vivado instead of Yosys (`vivado -mode batch -source digilent_genesys2.tcl`) along with the .tcl file necessary for Vivado.

Likewise, using this command

`~/FPGA/LiteX/litex-boards/litex_boards/targets/digilent_genesys2.py --toolchain=openxc7 --output-dir=experiment --build`

ends with this error message:
```
OSError: Unable to find or source Vivado toolchain, please either:
- Source Vivado's settings manually.
- Or set LITEX_ENV_VIVADO environment variant to Vivado's settings path.
- Or add Vivado toolchain to your $PATH.
```

I would expect it to create a shell script to invoke yosys and nextpnr-xilinx as provided by the openxc7 toolchain.

Mecrisp, 2024-12-11:
The same issue of the toolchain option being ignored, using Vivado as default instead, also happens with --toolchain=yosys+nextpnr and --toolchain=f4pga and also in combination with litex-boards/litex_boards/targets/xilinx_kc705.py

trabucayre, 2024-12-14:
In fact `--toolchain` is a default argument added by `LiteXArgumentParser`, but since this one isn't uses at `BaseSoC` constructor nor at `Platform` level default value remains used.

trabucayre, 2024-12-14:
@Mecrisp could you try [this PR](https://github.com/litex-hub/litex-boards/pull/629) ?

Mecrisp, 2024-12-16:
@trabucayre Thank you, I tried the changes and the toolchain option is recognised properly now. I am now running into path and dependency issues, as I installed openXC7 with the recommended nix based installer instead of the snap one that the LiteX core expects; but regarding the Genesys2 target file, this is fine and the option is properly handled now. Please merge it as-is.