litex-hub/litex-boards #634: Numato Tagus PCIe pin constraints not working properly

davidv1992, 2025-01-03:
When building the Numato Tagus board target using
```
python3 numato_tagus.py --with-pcie --cpu-type None --build
```
the resulting bitstream doesn't function on device correctly (the pcie endpoint is not visible from the OS), and a dig through the IO report indicates why: The constraints for the PCIe port are not properly applied for some reason and instead of the intended pins (A4,B4,A8,B8) for the TX/RX pins of the PCIe x1 slot, the vivado toolchain uses the pins (C7, D7, C9, D9)

This was observed building using the Vivado 2024.1 software. I have gotten the numato pcie sample project working with the same software, so it seems unlikely to be a bug in Vivado.

If this is not the correct place to report this bug (which probably is likely, given that the board files themselves look fine to me) please let me know where I should report it.

enjoy-digital, 2025-01-14:
Hi @davidv1992,

sorry for the delay, that's the right place to report the issue yes and thanks for the feedback. The issue is probably similar to what is observed on the Mini baseboard/Acorn when configured for PCIe X1 where Vivado does a switch of the PCIe pins. To avoid this, you'll have to use constraints similar to [these](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/litex_acorn_baseboard_mini.py#L149-L150).

To find the correct GTPE2_CHANNEL, you can open your routed design checkpoint and use the floorplan editor to locate it based on the pinout.

enjoy-digital, 2026-10-21:
@davidv1992: The PCIe pins are on `GTPE2_CHANNEL_X0Y4` but the PCIe IP was placing lane 0 on `GTPE2_CHANNEL_X0Y7` (the C7/D7/C9/D9 pins you saw). #930 forces the right channel. A Vivado build now places the GTP on A4/B4/A8/B8, but I don't have the board, so could you test it if you still have one? Thanks!

closed 2026-10-21 by enjoy-digital