litex-hub/litex-boards #648: Support Kintex-7 Base C board

offnaria, 2025-02-21:
Hello.

I got a board called Kintex-7 Base C from [AliExpress](https://ja.aliexpress.com/item/1005005361609787.html) and ported LiteX onto it, successfully running the donut of the demo application as the first step. (Video: https://x.com/i/status/1891150920253133077)
I would like to push the initial support of the board.

However, since I've obtained an XDC file for the board from the supplier, I'm not sure whether it is allowed to disclose information, such as pin assignments.
I've visited the webpage of the developer (https://bochenjingxin.com/), but I couldn't find the board XDC.
Shall I contact the developer to ask if I can publish the information?

offnaria, 2025-03-25:
I emailed the developer, and they allowed to publish it.
I'll make a PR later.

enjoy-digital, 2025-03-26:
Hi @offnaria,

good, thanks! If the board vendor allowed it, happy to review and merge the PR yes.

enjoy-digital, 2025-05-05:
Thanks @offnaria for the https://github.com/litex-hub/litex-boards/pull/654 PR and @trabucayre for the review/merge. We can now close this since merged.

closed 2025-05-05 by enjoy-digital

tommythorn, 2025-05-05:
The AliExpress link doesn't work anymore (though you get a quick glimpse of what it was).  Kindly consider including a photo (and other collateral) in the repo.