litex-hub/litex-boards #656: Digilent Zedboard gnores --toolchain switch

berdav, 2025-04-09:
Hi.

I'm trying to compile a basic example for the Digilent Zedboard without using vivado toolchain and using yosys+nextpnr (or any other open source toolchain).

If I use any toolchain and the command
```
./digilent_zedboard.py --build --toolchain yosys+nextpnr --load
```

I get the following error:

```
OSError: Unable to find or source Vivado toolchain, please either:
- Source Vivado's settings manually.
- Or set LITEX_ENV_VIVADO environment variant to Vivado's settings path.
- Or add Vivado toolchain to your $PATH.
```

How can I specify an alternative toolchain from the vivado one for this board?

Is any other toolchain supported at all? I see the list of toolchains from --help:

```
ubuntu@fpga-open-source-tools:~/litex-boards/litex_boards/targets$ ./digilent_zedboard.py --help | grep toolchain
usage: {'description': 'LiteX SoC on Zedboard.'} [-h] [--toolchain {vivado,f4pga,yosys+nextpnr,openxc7}] [--build]
  --toolchain {vivado,f4pga,yosys+nextpnr,openxc7}
          FPGA toolchain (vivado or f4pga or yosys+nextpnr or openxc7). (default: vivado)
Vivado toolchain options:
```

trabucayre, 2025-04-09:
Indeed, currently `--toolchain` option is not propagated to the `Platform()`.
Please check/test this [PR](https://github.com/litex-hub/litex-boards/pull/657)

trabucayre, 2025-04-28:
@berdav I have merged the PR: now it's possible to uses zedboard with opensource toolchain

berdav, 2025-04-30:
Thank you so much @trabucayre ! :)

I'm closing the Issue, I will reopen it if I find some problems.

closed 2025-04-30 by berdav

trabucayre, 2025-04-30:
Just a follow up: I have [this commit](https://github.com/trabucayre/litex/commit/fcce2be9a985e8dcc181d26f947609b7e6fe4136) to allows building the zynq7000 with openXC7. It's a WIP.